boltgolt/howdy #409: Floating point exception

blaketrombone, 2020-08-02:
I am running Fedora 32. I've run "sudo howdy add", howdy asks for a name, then it says "please look straight into the camera" and the IR sensor blinks. then howdy returns "Floating point exception" and quits

[image]

boltgolt, 2020-08-07:
This is an issue with OpenCV, you could try reinstalling that and it's python bridge

tomoshokas, 2020-08-07:
I'm in the same situation, hence it worked until one day I run update of system. 

I have dug deeper ;) by running coredumpctl and got following debug output:

Core was generated by `/usr/bin/python3 /usr/bin/howdy add'.
Program terminated with signal SIGFPE, Arithmetic exception.
#0  0x00007f82bcb5588f in sgemm_thread_tn (args=0x7fff37af4c50, range_m=0x0, range_n=0x0, sa=0x7f8296000000, sb=0x7f8296064000, mypos=<optimized out>) at level3_thread.c:759
759	    nthreads_n = (n + SWITCH_RATIO * nthreads_m - 1) / (SWITCH_RATIO * nthreads_m);
[Current thread is 1 (Thread 0x7f82da9d0740 (LWP 31604))]

The formula comes from /usr/lib64/libopenblas.so.0  vesrion 0.3.9-3, no downgrade possible through dnf.

ldepandis, 2020-08-12:
> The formula comes from /usr/lib64/libopenblas.so.0 vesrion 0.3.9-3, no downgrade possible through dnf.
> 
> Re-installation of OpenCV and python3-opencv does not help.
Actually you can downgrade openblas to 0.3.9-2 but still it does not solve the situation, just like it doesn't by downgrading opencv and python3-opencv.

s256, 2020-09-05:
Same here. Fedora 32.

simonerlic, 2020-09-06:
Same issue here, also Fedora 32. Any ideas on a workaround?

dswij, 2020-09-06:
This copr repo works for me instead of the one from readme, it also has a clear instruction there.

[https://copr.fedorainfracloud.org/coprs/principis/howdy/](https://copr.fedorainfracloud.org/coprs/principis/howdy/)

s256, 2020-09-06:
> This copr repo works for me instead of the one from readme, it also has a clear instruction there.
> 
> https://copr.fedorainfracloud.org/coprs/principis/howdy/
> 
> I have also tried running python directly from the latest src/cli.py and it's working fine.

Confirm working for me too. Thank you!

simonerlic, 2020-09-06:
> This copr repo works for me instead of the one from readme, it also has a clear instruction there.
> 
> https://copr.fedorainfracloud.org/coprs/principis/howdy/
> 
> I have also tried running python directly from the latest src/cli.py and it's working fine.

Can also confirm working! Thanks!

dswij, 2020-09-07:
@boltgolt Since it seems that this fixes the issue for the newer fedora, could we add this alternative repo to the readme? I think I saw the maintainer here on some other issue @principis

principis, 2020-09-07:
@dswij I'd rather not actually. I'm quite new at packaging (It's mostly just trial and error), and I don't think I have the time to quickly fix other people's issues. I do use it myself, so I'll keep maintaining it so that it at least works for most people.

albfan, 2020-09-10:
I have arch linux and fedora. https://copr.fedorainfracloud.org/coprs/principis/howdy/ solve the floating point exception (probably because it uses howdy 2.6.1 ? not sure) arch linux is on 2.6.1 too.

But the instructions to setup selinux are mandatory on fedora (`unknown error: 1` is shown if not configured)

@principis  I would just add that instructions on principis are only for kde. I setup `/etc/pam.d/login` and it works for gdm (GNOME desktop)

```
auth     [success=done ignore=ignore default=bad] pam_selinux_permit.so
auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
```

principis, 2020-09-11:
@albfan the floating point exception has nothing to do with howdy 2.6.1 I suspect dlib.

Thank you, I'll change it! :)

albfan, 2020-09-11:
@principis just that /etc/pam.d is little complex so any guidance on other desktop is great.

On arch Linux system-local-login is enough, but on fedora there's no such script, so some guidance would help. I guess login works, but maybe is not the exact best location

closed 2026-11-16 by boltgolt