boltgolt/howdy #411: Option to skip Howdy if another monitor is attached

alexanderadam, 2020-08-04:
I guess the title says it all.
I'm not looking to the IR & camera on the notebook when I have an external monitor attached.

boltgolt, 2020-08-07:
While this would be a good addition, it would add an additional dependency that i'm not sure a lot of people use. Maybe it's possible to make a bash script that takes input from the `xrandr` and feeds it into `sudo howdy disable`?

mikk150, 2020-11-23:
you should be able to read this information from `/sys/class/drm`, it should work with wayland and X11

closed 2021-04-02 by boltgolt

alexanderadam, 2021-04-02:
Okay, but I'm still not sure what to do now.
So how/where exactly should I use that information? I'm not even sure how howdy is integration in which parts of the system. :shrug: