boltgolt/howdy #415: Rotate camera image

HughPH, 2020-08-09:
So here's the thing: My camera is hanging underneath my monitor. That means when I'm in a conference call, my eyes are looking above the camera instead of below it, which means they look more open. All the conference call software and recording software I use can flip & rotate as necessary. Unfortunately Howdy does not - and I couldn't find anything in the config for that either.

boltgolt, 2020-08-10:
This could be implemented, but might be feature creep

Doomsdayrs, 2020-08-20:
feature creep?

boltgolt, 2022-01-16:
Very good progress in #627

boltgolt, 2026-11-16:
Hey, thanks for reporting this! Closing this as a duplicate of #507, which asks for the same thing (rotating the camera image 180 degrees). Let's keep the discussion there.

closed 2026-11-16 by boltgolt