boltgolt/howdy #418: Dell XPS 13 (9300) - "Face Detection Image Too Dark" error during daylight

exhile, 2020-08-12:
So when I turn on my XPS 13 and go to the Ubuntu login page, the IR camera and LED light turn on but fail to log me in as the error, "Face detection image too dark" error message appears. I have to type in my password to login. The lighting conditions are fine as it is day time. I had no previous problems before.

I do a 

`sudo howdy config`

I type in my password because "howdy" doesn't work and have edited the "dark_threshold value from 50 to 65, 75 and 80 with no solution to the problem.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04.1

Howdy version: 2.6.0

freudFlintstone, 2020-08-17:
I'm experiencing the same issue. Running `sudo howdy test` shows images at 15 FPS, looking fine, and a **dark frame** warning at all times. I be happy to help with diagnostics.
I'm using **dark_threshold** set to 90 and can login some of the time. I noticed that tilt my head vertically slightly during reading seems to improve success rate, probably because I'll certainly move into the same position of the model at some point.

freudFlintstone, 2020-08-17:
@exhile 
I set **use_cnn = false**, **dark_threshold = 90** and clear the models, then added 5 models, each in slightly different positions. It now works almost 100% and much faster. I tried installing python3-opencv and recording the models again with **use_cnn = true** but it didn't make any difference.

boltgolt, 2020-09-02:
@freudFlintstone has the right solution. "too dark" can also occur if there is a bright light behind you, blinding the camera.