boltgolt/howdy #419: Unable to get howdy to start for verification

bob8677, 2020-08-13:
When I run sudo, howdy does not start even though I enabled it in the pam.d file for sudo, which should be all I have to do, according to both the wiki in this repo as well as the [arch wiki](https://wiki.archlinux.org/index.php/Howdy#Configuration)

output of cat /etc/pam.d/sudo
> #%PAM-1.0
auth            sufficient      pam_python.so /lib/security/howdy/pam.py
auth            include         system-auth
account         include         system-auth
session         include         system-auth

`sudo howdy test` works fine and I was able to add a user model just fine, but nothing shows up.

I've also run sudo python2 /lib/security/howdy/pam.py and no errors show up, although I'm not sure that means anything.

I am also unable to find the log file, which should be /var/log/auth.log as far as I know.

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): arch

Howdy version: Howdy 2.6.0

hissssst, 2020-08-16:
Experiencing the same on Arch
```
Howdy 2.6.0
```

hissssst, 2020-08-16:
Fixed this issue with
https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types

bob8677, 2020-08-16:
I had done this before I think.
>output of cat /etc/pam.d/sudo
 #%PAM-1.0
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth include system-auth
account include system-auth
session include system-auth

Which should be all I need as far I as I am aware, but it still does not start.

boltgolt, 2020-09-02:
Sounds like pam-python isn't working correctly, does running `sudo python2 /lib/security/howdy/pam.py` return a status code 1?

shivindera, 2020-09-08:
> Sounds like pam-python isn't working correctly, does running `sudo python2 /lib/security/howdy/pam.py` return a status code 1?

Having the same problem. Running the above returns nothing.

boltgolt, 2020-09-09:
It shouldn't return anything, i'm asking for the status code. You can run `echo $?` after you ran the python command

shivindera, 2020-09-09:
> Could you also try to run `sudo python3 /lib/security/howdy/compare.py`?

I ran the above commands. See the attached image.

[image]

boltgolt, 2020-09-09:
Sorry my fault, second command should be `sudo python3 /lib/security/howdy/compare.py USERNAME` where you should replace USERNAME with your actual username

shivindera, 2020-09-09:
When I run the above command it starts the IR camera verifies my face and then exits. If I run `echo $?`, it returns 0. 

If I timeout the verification, `echo $?` returns 11.

boltgolt, 2020-09-28:
Seems like it's working exactly as it should, except when called by PAM. Could you set more broad permissions on the howdy folder? For example `chmod -R a+x /lib/security/howdy`

bob8677, 2020-10-22:
I just got howdy working, after not trying for a while, what I think happened is that I had used the wrong pam package for python, the one from the aur worked, not from pacman or pip. I had to remove the dependencies from the pkgbuild to get to work for some things that I had installed with pip, but I think I just had the wrong pam package.

closed 2020-10-22 by bob8677

killi1812, 2024-05-21:
Hey @bob8677 could you post exact packages an versions that you use