boltgolt/howdy #422: pam_python.so missing on Fedora

Swiftb0y, 2020-08-17:
I'm using howdy from copr from @luyatshimbalanga and I was able to get it running with some minor patches but I've now discovered that fedora is simply missing pam_python.so
I would appreciate if you could provide the missing pam_python library in the copr repo.

Linux distribution (if applicable): Fedora 32

Howdy version: 2.6.0

luyatshimbalanga, 2020-08-18:
pam_python is already packaged on https://copr.fedorainfracloud.org/coprs/anaumov/pam-python/.
What will be great is pam_python.so getting ported to python 3 instead. See the review process on https://bugzilla.redhat.com/show_bug.cgi?id=1641386

Swiftb0y, 2020-08-18:
Thanks for your response. It would probably be wise to add that info to the fedora install instructions. Unfortunately, I struggle to enable that package. Do you know any workarounds?
```
sudo dnf copr enable anaumov/pam-python
[...]
Error: This repository does not have any builds yet so you cannot enable it now.
```

LukaJankovic, 2020-08-19:
I ran into the same issue, but managed to get pam_python on fedora using the rpm sphere repo, however now instead I get an error about the dlib module not being found... despite installing manually / with pip2 / pip3, etc., which is strange considering it loads correctly from a python shell (both 2 and 3).

Swiftb0y, 2020-08-19:
dlib is packaged with the copr repo. try installing `python3-dlib` from the repo instead of pip. when you do install dlib via pip from as your user, it will only be available for that user but howdy is run as root so it won't be able to find that version of dlib.

LukaJankovic, 2020-08-19:
Tried installing `python3-dlib` from the repo, but now i get the error 

    No module named '_dlib_pybind11'

    Can't import the dlib module, check the output of
    pip3 show dlib

and they can't be found by the python shells despite being reported working by pip and being installed in the correct locations, `/usr/lib64/python3.8/site-packages` that is.

Having dlib installed using `sudo pip3 install dlib` is the way I had it set up before and results in howdy working however whenever a sign in is initiated I get the error

    Traceback (most recent call last):
      File "/lib64/security/howdy/compare.py", line 18, in <module>
        import dlib
    ModuleNotFoundError: No module named 'dlib'
    Unknown error: 1

And again, it works in the python shell.

principis, 2020-09-02:
@LukaJankovic I've created my own builds because of this (and other problems). Maybe it'll will solve your problems https://copr.fedorainfracloud.org/coprs/principis/howdy/

dbdorman, 2020-10-17:
@principis Thanks for your copr repo; do you plan to update it for Fedora 33? It's not installing for me on Fedora 33. I also tried compiling myself, and while howdy works for me, pam_python does not (I get a segmentation fault when trying to run it after compiling it myself). Any clues?

principis, 2020-10-20:
@dbdorman I'll look into it

dbdorman, 2020-10-24:
@principis Thanks, I was able to install from your copr build, but still getting a segmentation fault when executing pam_python, so must be something with my system. I'll try debugging

socketpair, 2024-07-24:
So what is the result  of everything that ?

principis, 2024-07-24:
> So what is the result of everything that ?

Could you please elaborate?

socketpair, 2024-07-24:
@principis 
https://github.com/rpmsphere/spec/pull/4
But I don't know how to add .patch there. I managed to compile it for Fedora39, but did not test yet.

principis, 2024-07-24:
> @principis [rpmsphere/spec#4](https://github.com/rpmsphere/spec/pull/4) But I don't know how to add .patch there. I managed to compile it for Fedora39 (i.e. Python 3.12), but did not test yet.

I have no idea what you are trying to accomplish. I have packaged howdy [here](https://copr.fedorainfracloud.org/coprs/principis/howdy/), which should work fine.

closed 2026-11-16 by boltgolt