boltgolt/howdy #424: Howdy on GPU

gabrielspadon, 2020-08-27:
Hi Folks,

I have been digging around trying to run Howdy on GPU, more specifically the CNN model.
I checked all the issues and tutorials here but didn't find anything that helped me do it.
I'm running PopOS! 20.04 on an XPS 17 (2020) with CUDA 10.2 (ver. 440.100).

So, I did the following:

* Installed Howdy from the original repository;
* Uninstalled OpenCV 4.4 that comes with it; and,
* Compiled my own OpenCV 4.4 with CUDA and CUDNN enabled and installed it on the system-level Python (as Howdy runs with sudo).

Howdy works just fine, as it should be - but not on the GPU.
When I test the OpenCV on Python, I can check that CUDA is working.
However, no matter what I change in Howdy Config, I never see it using any GPU on my nvidia-smi (using `sudo howdy test`).

Here is my OpenCV installation script:

```bash
sudo apt --reinstall install build-essential checkinstall cmake doxygen gfortran git libatlas-base-dev libavcodec-dev libavformat-dev libavresample-dev libdc1394-22 libdc1394-22-dev libeigen3-dev libfaac-dev libgflags-dev libgoogle-glog-dev libgphoto2-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgtk-3-dev libhdf5-dev libjpeg-dev libmp3lame-dev libopencore-amrnb-dev libopencore-amrwb-dev libpng-dev libprotobuf-dev libswscale-dev libtbb-dev libtheora-dev libtiff-dev libv4l-dev libvorbis-dev libx264-dev libxine2-dev libxvidcore-dev pkg-config protobuf-compiler python3-dev python3-pip python3-testresources unzip v4l-utils x264 yasm -y

sudo ln -s -f /usr/include/libv4l1-videodev.h /usr/include/linux/videodev.h

mkdir -p OpenCV && cd OpenCV

wget -O opencv.zip https://github.com/opencv/opencv/archive/4.4.0.zip
wget -O opencv_contrib.zip https://github.com/opencv/opecnhv_contrib/archive/4.4.0.zip

unzip opencv.zip
unzip opencv_contrib.zip

cd opencv-4.4.0
mkdir -p build && cd build

cmake \
-D CMAKE_BUILD_TYPE=RELEASE \
-D CMAKE_C_COMPILER=/usr/bin/gcc-8 \
-D CMAKE_INSTALL_PREFIX=/usr/local \
-D INSTALL_PYTHON_EXAMPLES=ON \
-D INSTALL_C_EXAMPLES=OFF \
-D WITH_TBB=ON \
-D WITH_CUDA=ON \
-D BUILD_opencv_cudacodec=OFF \
-D ENABLE_FAST_MATH=1 \
-D CUDA_FAST_MATH=1 \
-D WITH_CUBLAS=1 \
-D WITH_V4L=ON \
-D WITH_QT=OFF \
-D WITH_OPENGL=ON \
-D WITH_GSTREAMER=ON \
-D OPENCV_GENERATE_PKGCONFIG=ON \
-D OPENCV_PC_FILE_NAME=opencv.pc \
-D OPENCV_ENABLE_NONFREE=ON \
-D OPENCV_PYTHON3_INSTALL_PATH=/usr/lib/python3/dist-packages \
-D OPENCV_EXTRA_MODULES_PATH=~/OpenCV/opencv_contrib-4.4.0/modules \
-D PYTHON2_EXECUTABLE=/usr/bin/python3.8 \
-D BUILD_EXAMPLES=ON \
-D WITH_CUDNN=ON \
-D OPENCV_DNN_CUDA=ON \
-D CUDNN_LIBRARY=/usr/lib/cuda/lib64/libcudnn.so.7.6.5 \
-D CUDNN_INCLUDE_DIR=/usr/lib/cuda/include  \
-D CUDA_ARCH_BIN=7.5 ..

make -j16

sudo make install && sudo ldconfig
sudo chmod 770 -R /lib/security/howdy
sudo chown -R root:$USER /usr/lib/security/howdy/
```

Here is my Howdy `config.ini`:

```bash
[core]
detection_notice = false
no_confirmation = true
suppress_unknown = true
ignore_ssh = true
ignore_closed_lid = true
disabled = false
use_cnn = true

[video]
certainty = 3
timeout = 4
device_path = /dev/video0
max_height = 480
frame_width = -1
frame_height = -1
dark_threshold = 80
recording_plugin = opencv
device_format = v4l2
force_mjpeg = false
exposure = -1

[snapshots]
capture_failed = true
capture_successful = true

[debug]
end_report = false
```

Any ideas on how to do it?

nalin29, 2020-08-28:
I also have opencv built with cuda enabled and can't seem to get howdy to use gpu acceleration.

boltgolt, 2020-09-02:
Weird, using `use_cnn` should enable GPU usage. I did not write that part of the code so more investigation is needed

gabrielspadon, 2020-09-07:
> Weird, using `use_cnn` should enable GPU usage. I did not write that part of the code so more investigation is needed

Let me know if you need more information about my setup.

Danhyal-Masood, 2020-09-24:
afik all you need is dlib compiled with cuda support, atleast for me that is all I needed

wmmc88, 2020-09-26:
> afik all you need is dlib compiled with cuda support, atleast for me that is all I needed

I initially tried recompiling opencv4 from source with cuda support and howdy looked like it wasnt using it. Howdy now uses gpu after I reinstalled dlib with cuda.

```
sudo -i
pip3 uninstall dlib
pip3 install dlib --verbose --no-cache-dir # You should see the CMake logs and you should see it detecting CUDA and it saying that dlib will be built with CUDA
```

You can verify it worked:
```
sudo -i
python3
>>> import dlib
>>> dlib.DLIB_USE_CUDA
True
```

When you run `sudo howdy test` you should see python3 using gpu mem in `nvidia-smi -l 1`

gabrielspadon, 2020-10-08:
Getting DLib to work with CUDA on PopOS was tricker than just installing from pip.
But now, Howdy is using my GPU when making face recognition with CNNs.
Thanks, @wmmc88 .

closed 2020-10-08 by gabrielspadon

kylebakerio, 2022-11-14:
> Getting DLib to work with CUDA on PopOS was tricker than just installing from pip. But now, Howdy is using my GPU when making face recognition with CNNs. Thanks, @wmmc88 .

this is very teasing; I know this was a while ago, but do you happen to recall / are you willing to share what was required...?