boltgolt/howdy #428: Use howdy without IR

devmrfitz, 2020-08-31:
Is it possible to use howdy with a normal webcam (without IR support)? I understand that IR is essential in insufficient light, but if  the lighting is bright enough, can I use Howdy on a normal webcam?

mzzntn, 2020-09-01:
Yes you can it works perfectly even without IR but, as you can imagine the level of security will be lower

amirkaveh, 2020-09-02:
You can use it either with or without an IR camera. Set your desired camera in the config file using `sudo howdy config` and write a camera path in front of device_path.
You can find your camera paths using the "v4l-utils" package and with the `v4l2-ctl --list-devices` command.

boltgolt, 2020-09-02:
You can but it's not a great solution. Just a picture of you on a phone will probably fool Howdy when using a normal webcam.

closed 2020-09-02 by boltgolt

devmrfitz, 2020-09-02:
> You can but it's not a great solution. Just a picture of you on a phone will probably fool Howdy when using a normal webcam.

Can an "eye blink detection" be integrated somehow?

boltgolt, 2020-09-02:
Possibly, if you want you can open a new issue for that