boltgolt/howdy #429: Where does the beeping sound come from ? Can I make it quieter ?

ozls, 2020-09-01:
Hi,

I am using Howdy and I love it. It works perfectly. This is not really an issue nor a feature request, but I am getting a beeping sound each time Howdy starts and authenticates me, so two times. Its useful I guess, but it's also very loud and I would like to tone it down / replace it with a notification or something. Where does that sound come from ? Does anybody else have this ?

I am using Manjaro with Linux kernel 5.7.15 and Howdy 2.6.0 from the AUR. Thanks in advance !

ozls, 2020-09-01:
Closed because I used a trick to disable this speaker beeping altogether. But I'm still interested in how Howdy makes the speakers beep.

closed 2020-09-01 by ozls

boltgolt, 2020-09-02:
I'm thinking the beeping was caused by Howdy accidentally sending a emergency broadcast on every authentication prompt, this should be fixed in 2.6.1

tarun-varier, 2020-09-06:
@ozls what trick did u use tho to disable that annoying sound?
xset b off doesn't make no differencce

ozls, 2020-09-06:
The annoying, full volume beeps come from the `pcspkr` module (on my Manjaro install at least). I think what I did was `echo "blacklist pcspkr" | tee /etc/modprobe.d/nobeep.conf`. I found it [here](https://wiki.archlinux.org/index.php/PC_speaker)

tarun-varier, 2020-09-06:
@ozls thanks a ton...you are a messiah