boltgolt/howdy #431: Works for sudo not for login

SalmonSeasoning, 2020-09-01:
**Everything is up to date.** I have multiple faces added (one plain, one with glasses, one with headphones, one with both) and it works fine and dandy every single time I've tried going to terminal and running `sudo -i`, but it times out every single time no matter what I do when I am at the login prompt.
I followed this tutorial: [https://itsfoss.com/face-unlock-ubuntu/](https://itsfoss.com/face-unlock-ubuntu/)

Haven't found any similar issues, nor any possible resolutions.

Linux distro: Pop!_OS (Latest LTS)

Howdy version: 2.6.0

Here is the minified config file:

> [core]
> detection_notice = false
> no_confirmation = false
> suppress_unknown = false
> ignore_ssh = true
> ignore_closed_lid = true
> disabled = false
> use_cnn = false
> [video]
> certainty = 2.8
> timeout = 4
> device_path = /dev/video0
> max_height = 320
> frame_width = -1
> frame_height = -1
> dark_threshold = 50
> recording_plugin = opencv
> device_format = v4l2
> force_mjpeg = false
> exposure = -1
> [snapshots]
> capture_failed = true
> capture_successful = true
> [debug]
> end_report = false

boltgolt, 2020-09-02:
Could you try upping your `certainty` to something higher, like 5?

olssy, 2020-09-02:
I had this issue as well and this fixed it:
sudo chmod o+x /lib/security/howdy/dlib-data/

https://github.com/boltgolt/howdy/issues/238#issuecomment-517842301

SalmonSeasoning, 2020-09-02:
Okay @boltgolt  I got it to work finally. I'm not sure what the problem was because when I ran @olssy's command, it didn't immediately recognize me as it had the day before. When I upped the certainty to 5 it recognized me, and also let me log in through the lock screen.

So thanks for helping me. I still find it weird that it worked on `sudo -i` but not on the lock screen.

If you need any more information from me, go ahead and email me at tyqualters (at) pm (dot) me, and I'll do what I can.

closed 2020-09-02 by SalmonSeasoning