boltgolt/howdy #433: Eye blink detection

devmrfitz, 2020-09-03:
Is it possible to detect if the face in front of the camera is blinking his/her eyes? This will really help to mitigate simple loopholes like using a life-size image of a person, especially when camera lacks IR.

boltgolt, 2020-09-03:
Possibly, but this would be very hard to get right. 

Howdy only really gets about 10 to 20 FPS on most PCs I've seen, and a blink lasts for about 100ms. I believe this only gives us 1 or 2 frames in which we have to detect the blink. I don't think this is feasible 

Besides, adding the detection of facial features (like the eyes) requires another pass in the face detector, lowering FPS even more 

It might be possible to ask a user to do something more easier to detect, like close the left eye. I would love feedback on this and any other ideas or comments are very welcome

Amarjit3, 2021-03-18:
I think this is a really good idea. In combination with IR, the face unlock would become very robust

fatkaratekid, 2021-10-03:
I think I can work on this one. It could go one step further - checking if a person is real vs image.

fatkaratekid, 2021-11-17:
@boltgolt 
Is there a requirements file somewhere? I tried simply searching amongst files, but no luck.
What is the lowest version of python howdy supports?

boltgolt, 2021-11-17:
Python 3.5.0 should be the lowest supported

bhavyammodi, 2023-02-09:
Windows hello turns on the normal camera as well (I know it from the fact that the white light is on in windows and not on Linux howdy), normal camera can help detect closed/blinking eyes I guess.