boltgolt/howdy #435: Brighter Image Overexposed

dav-is, 2020-09-03:
It seems like the brighter IR image is being overexposed. The darker image seems just about right.

[image]
[image]
How is it possible one of these failed and the other was successful... they both look pretty close to me

Here, the brighter image looks about right, but the darker images are barely exposed.
[image]

Maybe we need a few moments of low exposure, then a few moments of high exposure (sort of like how HDR works)

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04

Camera: Logitech Brio

Howdy version (`sudo howdy version`): 
```
$ sudo apt list howdy
Listing... Done
howdy/focal,focal,now 2.6.1 all [installed]
```

dav-is, 2020-09-03:
It looks fine when using `sudo howdy test` but it seems pretty unreliable and times out when recognizing outside of testing. The only way I've seen to debug is using these screenshots

eskander12, 2022-04-08:
but can you not just make it so it takes a bit longer before the picture is saved

boltgolt, 2022-04-08:
You're replying to a 2-year-old issue but in short it's normally good for diagnosing issues to save the first few frames

closed 2022-04-08 by boltgolt