boltgolt/howdy #436: Add Option to Make Device Optional

dav-is, 2020-09-03:
I'm using my laptop away from my external webcam (Logitech Brio). I'd like an option that silences this error and just prompts for my password.

```
$ sudo -i
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7fde469518b0>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
Unknown error: 1
[sudo] password for davis: 
```

arifer612, 2020-11-11:
I think that this is similar to #411 and it has been filed under the _wontfix_ label

dav-is, 2020-11-13:
@arifer612 I disagree. That issue seems to have been filed under `wontfix` because "it would add an additional dependency that i'm not sure a lot of people use". This would be a flag in configuration so it doesn't throw an error when the camera was not found. There's no need for any new dependencies.

boltgolt, 2020-12-03:
PR #455 was just merged and does exactly what you proposed, you'll find it in the next release!

closed 2020-12-03 by boltgolt