boltgolt/howdy #437: Add refresh button to login UI

MyCatCondo, 2020-09-03:
I think it would be useful to add a refresh button to the logon UI so when it says face not recognized at the login prompt you can click a refresh icon and it will scan again.

Batcastle, 2020-09-04:
I second this request, however, I think making a way to control this via CLI would be better so different display managers can work with it. Furthermore, it could be used from a keyboard shortcut while authenticating `sudo`. In which case, I think needing to be able to control this from a separate process would be ideal.

I would also like to include a request for a command to halt facial recognition and fall back to password only, in case someone other than the user Howdy is set up for needs to log in and has the password to do so. This way, they are not caught as a failed login attempt. If this additional request should be made into a separate issue, please let me know and I will do so.

MyCatCondo, 2020-09-04:
Yeah, I’m using Debian (Parrot OS to be exact) and it would be useful to have it on all compatible desktop environments. But for the login screen (PAM) on Debian it would be useful to get a refresh icon or something right next to the desktop environment switcher button. For the other desktop environments, like BunsenLabs Linux (it’s also Debian, and it doesn’t work with Howdy in the lock screen), maybe a keybind to trigger the scan would work. If course you could control all of this with the `howdy` command.

I agree with you about the keyboard shortcut. Maybe Alt-Shift-R might work if it isn’t already a keybind for the OS.

Maybe Howdy shouldn’t scan automatically on the login screen, and it should only scan on the login screen if you click the little refresh icon. That could be an option in the configuration file. The refresh icon could also be an eye or a smiley face like Windows Hello. It could also be the Howdy icon. For the command, a keybind may also work, like Ctrl-Shift-D maybe if it isn’t a keybind already.
Yeah, that may be good for a second feature request. You can create another one for that if you want.

Batcastle, 2020-09-04:
Well the reason I mention specifically a keyboard shortcut and not the login UI is that in order to implement that, it would take work on many display manager's parts. This would cause it to work perfectly for some users, while others would have a broken, if even usable, experience. We want to prevent that, so by going through the command line, the display manager option is still there, but it also makes the keyboard shortcut an option, at the discretion of the user or distro maintainer. Setting that upon installation would be bad practice.

But, I that's all besides the point. It seems this is at least somewhat possible. I noticed when attempting to sign into my laptop with Howdy earlier that after failed to recognize my face for whatever reason, and I got my password wrong (the pain of über long passwords), LightDM attempted to use Howdy to authenticate my face again. Upon further investigation, `sudo` does the same thing. This leads me to believe this is possible. It may require some interaction with PAM though.

My theory is that PAM cycles through each authentication method a given number of times. This looks like just asking a user for a password a few times normally, but when multiple authentication methods are possible, this behavior I observed is what we get. Essentially, its a minor bug we could take advantage of. I'm going to look into getting PAM to restart it's authentication cycle and see if there is a way to do it. If so, adding this functionality should be fairly easy.

boltgolt, 2020-09-04:
This would be nice but is difficult to implement. Howdy has very little control of the UI right now, everything is done through the PAM API. A refresh button isn't possible within that API. 

I'm working on a stand alone UI which could include this

MyCatCondo, 2020-09-04:
Yeah maybe a keyboard would work instead of a refresh button.

PJSingh5000, 2021-01-12:
I agree with boltgot, this feature would be the responsibility of the display manager (GDM, LDM, etc.) to implement.

There may be different ways of doing this from a UI/workflow perspective...

In GDM, if authentication fails, the password box is displayed along with the error message. Next to the password box is a "<" back button. Manually clicking this and then pressing the enter key makes Howdy attempt to authenticate again. However, this takes multiple steps.

One approach might be for GDM to display the password box and error message for 5 seconds (or whatever time is appropriate). If the user doesn't enter a password, GDM would automatically switch back to the user list. In this way, the user would only have to wait 5 seconds and press the enter key. In this way, waiting a few seconds and pressing the "Enter" key causes a "refresh" to happen, approximating the behavior you are looking for.

The other approach, of course, is placing a refresh button next to the password box and error message, as your are suggesting. However, this is a little misleading, because clicking the refresh button would force re-authentication (via howdy), but the user has not entered anything in the password box. This is misleading because visually it seems as if the user has chosen a "blank" password. (Of course we know that Howdy is using the camera to authenticate, but it still results in a confusing UI flow).

The ideal situation would be for the Display Manager to know there is an alternative authentication mechanism (Howdy) and ~not display a password box. Instead, it should have a refresh button, and an option to "Enter password". Clicking the Refresh button would re-trigger facial recognition, but if the user really wanted to type his/her password, they could click the "Enter password" option, type their password, and press Enter.

There may be other approaches as well. 

As you can see, this really needs to be a capability incorporated into the Display Manager.

I suggest opening an enhancement request with GDM, LightDM, etc.

boltgolt, 2021-01-12:
Even beter, PR #484 _might_ allow us to do password authentication and Howdy auth at the same time, and we could add an refresh feature in the future as we will be in more in control of the PAM flow

Batcastle, 2021-01-12:
I think adding a refresh feature, at least on command line, would be a smart idea. I agree making it the responsibility of the DM is the right course of action. However, making this feature easily implementable by giving it a CLI option instead, so that it can be implemented in that DM or essentially anything else from a keyboard shortcut, to a script a user creates, to `pkexec`'s backend, would make this feature more versatile.

closed 2021-04-02 by boltgolt