boltgolt/howdy #438: Unlock the login keyring

amirkaveh, 2020-09-04:
Hi. 
Everything works smoothly in Howdy version 2.6.1 on Ubuntu 20.04.1. 
The only thing is that when you log in to the system using Howdy for the first time, the OS prompts for your password because of the keyring. 
I think solving it is a tough one because of OS security rules, but not prompting for a password improves the experience very much. 

[image]

boltgolt, 2020-09-04:
Unfortunately this is unfixable at the moment, the keyring is secured with your password and without this password there is no way to unlock it. It's unfortunately a side effect of how the keyring works

closed 2020-09-04 by boltgolt

Nos78, 2022-12-14:
@boltgolt may I ask for your thoughts on the logic here? If this keyring is auto-unlocked when a user enters their login password from the login-screen, why can it not also be unlocked when a biometric login is provided? I assume the keyring password and the user password are one-and-the-same, else the same logic you state would also prevail?

Is there is no configuration we can change to facilitate this? For example, enabling auto-login (IE logging in without a password) will also repreduce this exact behaviour in prompting the for the password. So, continuing this logic flow, we're basically saying that the howdy tool is a glorified auto-login, thats more secure than the default passwordless user auto-login?

In summary, are you saying that for a face login to unlock the keyring, then the keyring tool needs modifying?

boltgolt, 2022-12-14:
Your first paragraph makes no sense to me. Your password is the encryption key to your keyring and you do not enter it when you log in with Howdy. There is no way around that specific issue. 

If you'd like you could set the keyring password to an empty string, but that does mean your keys are saved on disk in plain text.

barbalex, 2023-11-14:
> Your first paragraph makes no sense to me. Your password is the encryption key to your keyring and you do not enter it when you log in with Howdy. There is no way around that specific issue.
>
> If [...]

If this is the case, then howdy is pretty much useless. As @Nos78 mentioned, howdy is useless if it can not be used instead of entering a password.

I installed it and loved how it worked - until I realized this issue. So uninstalling again.

tokox, 2023-11-14:
I agree, but there is no way howdy can unlock that. Keyring is encrypted using your password and it can't be decrypted differently than entering password. You could reconfigure your keyring tool to work somehow different so no password will be needed (you will have to think about replacement) or for example create a script that will unlock the keyring after login from saved somewhere password or something like that. The second solution is only a very not recommended and unsafe workaround.

Nos78, 2023-11-14:
Oh wow this thread is a blast from the past! @boltgolt thanks for that answer—IIRC the password being used as the encryption key was the bit of the logic I'd overlooked; the first paragraph, in my original comment you responded to, that made no sense to you, was in reference to me being prompted for my Linux user password (or keyring password? I'm replying quickly here) when face recognition failed. And pondering the whole process, lead me to wonder why I could not "unlock" my keyring with a face recognition. 
The answer was simple, and your reply helped. I never replied and I apologise. Since this has been added to, I now wish to do so!

I guess I'm not alone in wanting howdy to completely replace password entry. However, that doesn't make it (howdy) useless—it's only in accessing a keyring that it gets....messy... and I now understand why. The passwd is the encryption key. Therefore impossible.

I do like blasts from the past!

boltgolt, 2023-11-14:
It's an issue basically all alternative single-factor PAM modules run into, if the keyring starts widely supporting TPMs then it would be a different story 

In its current state it's at least very useful to me personally, my laptop as of right now has 100+ days of uptime and I only had to type my password when I booted it. Since then Howdy has handled all my sudo prompts

Nos78, 2023-12-01:
> It's an issue basically all alternative single-factor PAM modules run into, if the keyring starts widely supporting TPMs then it would be a different story
>
> In its current state it's at least very [...]

Totally agree... it was just that one area that it became, less than useful. A unique use case, otherwise like you, I only need to use password when my appearance changes significantly. Like when I grew a beard through total laziness.

So much so, that when I needed to sign into my SMB share on a new PC. I almost content remember the darn password 😂

jeffshee, 2023-12-16:
@boltgolt Sorry if this has been asked before. Is there a way to configure PAM to avoid using face authentication for login if the keyring is locked (i.e., during the first bootup), similar to how Android, with biometrics enabled, requires a PIN/password after a reboot?

bochen87, 2024-01-15:
This is quite an issue for me - i tried using howdy, but due to keyring not getting unlocked, i have big issues:
- My VScode is stuck trying to signin to the various plugins
- Chrome is detecting my pc as a new pc and logs me out of everything
- slack can't login
- online accounts can't login

It's quite the pain. I had to unfortunately uninstall howdy due to this issue :-(

kizzard, 2024-11-01:
For anyone else running into this problem, it is possible to automate unlocking the keyring after login with https://codeberg.org/umglurf/gnome-keyring-unlock . The repository provides an Arch centric example for how to encrypt your password with the TPM and use it unlock the keyring upon login. Clevis didn't seem to work with the TPM in Pop OS, so I wrote another guide using to do a similar thing with tpm2-tools: https://gist.github.com/kizzard/166470fefe8fa64d2aa65e0235115318

popy2k14, 2024-11-02:
@kizzard thx a lot, that works like a charme on mint 22!

maksims-terjohins, 2024-11-28:
@kizzard Thank you for sharing your solution. Worked on my Ubuntu 22.04 with a slight change. I have shared my solution under your guide in the comment section.

MaxMansfield, 2025-03-04:
@kizzard Nice! Worked for me on Manjaro. I think using `openssl`  and `systemd` as mentioned in the comments really help polish the solution.

@boltgolt Is there any reason we can't add this solution to howdy as a option?