boltgolt/howdy #441: " 'VideoCapture' object has no attribute 'internal' " after trying add a new face model

atahan2000, 2020-09-11:
Got this exception error after trying to add a face.

"""
Exception ignored in: <function VideoCapture.__del__ at 0x7f338bb7ad30>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
"""

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2020-09-11:
Please search for other issues with you error before opening a new one

closed 2020-09-11 by boltgolt

epgeroy, 2022-10-06:
I have the same issue, any information related to this?

thanks in advance

boltgolt, 2022-10-06:
See #533