boltgolt/howdy #442: Face Unlock Not Working On Lock Screen In Elementary OS

prajul-L850, 2020-09-12:
I've installed Howdy today Only.When I try to use face unlock,the led indicator near the WebCam lightens up,but then nothing happens.After that,I tried to enter my password and log in,but it was frozen.Then I logged back in by rebooting the OS.
Please help me as soon as Possible.
I can't afford to reboot every time I put my PC to sleep.
 
Linux distribution (if applicable):Elementary OS

Howdy version :Howdy 2.6.1

prajul-L850, 2020-09-12:
eos Version: 5.1.7 Hera

grandaolionel, 2020-09-15:
I'm also on elementary OS Hera 5.1.7
Kernel version | 5.9.0-050900rc4
Howdy version | 2.6.1

Cannot reproduce the issue, Howdy works properly in the terminal and when booting/logging in/unlocking the session.

Hardware is Lenovo Yoga Slim 7 Ryzen 7 4800U

prajul-L850, 2020-09-15:
Well,I uninstalled Howdy the next Day

prajul-L850, 2020-09-16:
I installed it again today and now it works like a charm.I could unlock my pc as well as use howdy for the sudo operations in the terminal.Thank You Howdy,you've just simplified my life more.

closed 2020-09-16 by prajul-L850