boltgolt/howdy #445: INFRARED NOT WORKING

magest2002, 2020-09-15:
respected sir   
                      in windows hello facial recognisation  my laptop has 2 ir blaster which help in -recognisation i cant figure out how to turn on  plz help me 

thanks

magest2002, 2020-09-16:
hpspectre@pop-os:~$ v4l2-ctl --list-devices --all
Synaptics RMI4 Touch Sensor (rmi4:rmi4-00.fn54):
	/dev/v4l-touch0

HP TrueVision FHD RGB-IR: HP Tr (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1

Driver Info:
	Driver name      : uvcvideo
	Card type        : HP TrueVision FHD RGB-IR: HP Tr
	Bus info         : usb-0000:00:14.0-5
	Driver version   : 5.4.44
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV' (YUYV 4:2:2)
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=90 max=150 step=1 default=120 value=120
                           gain 0x00980913 (int)    : min=1 max=7 step=1 default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4000 value=4000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=10 max=626 step=1 default=156 value=156 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0

magest2002, 2020-09-16:
also im getting 2-7 fps on sudo howdy test

boltgolt, 2020-09-28:
Have you searched for other issues? This has come up a bunch of times i think

PJSingh5000, 2020-10-07:
I believe I am having the same/similar issue on my HP Spectre.

The command `v4l2-ctl --list-devices --all` returns the same information as above.

I am running Ubuntu 20.04 with latest updates.

```
$ sudo howdy add
Adding face model for the user XXXXXX
Enter a label for this new model [Initial model] (max 24 characters): XXXXXX
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7fb0fa8adee0>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
```

justitsi, 2020-10-28:
Hi @magest2002, I am running howdy on a HP Spectre x360 from 2017 (8th get intel), the IR emitters will only turn on for a specific video device (in my case it's `/dev/video2`) that can be set when running `sudo howdy config`. You can open something like cheese to see which video device is which