boltgolt/howdy #446: Able to install & add new faces; however, Howdy doesn't activate for authentication requests.

shubydoobydoo, 2020-09-16:
Hi - I'm on an XPS 2020 running Fedora 32, and was able to download Howdy after a few unsuccessful attempts (dlib file was missing, but after reinstall, everything worked). I had to go into sudo howdy config since my camera wasn't being detected - I discovered /dev/video0 is the IR camera for this version of XPS. 

I was also able to add a new face model and the test showed that everything was working. However, Howdy doesn't run whenever I have an authentication request (i.e. sudo -i or the log in screen). Does anyone have suggestions for what I should try next or has a sense of what the issue might be?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 32 Gnome 3.36.6 Wayland

Howdy version (`sudo howdy version`): Most recent version

dswij, 2020-09-16:
Might have something to do with pam

Have u looked at /etc/pam.d/ ?

shubydoobydoo, 2020-09-16:
Yeah, I went into /etc/pam.d/sudo and added 

auth       sufficient   pam_python.so  /lib/security/howdy/pam.py

Still doesn't launch Howdy whenever it's prompted for a password. Do you think there's another directory I should add that to?

btb23, 2020-09-17:
I solved the same issue by setting permissions...

# chmod -R a+r /lib/security/howdy``

shubydoobydoo, 2020-09-18:
That did it!!! Thank you both for all the help!

closed 2020-09-18 by shubydoobydoo