boltgolt/howdy #448: Sudo/GDM login working, not working on screen lock, hibernate, sleep (Manjaro KDE with GDM instead of SDDM)

mcwillzz, 2020-09-22:
As the title says, I have Howdy working to log me in at the GDM login screen and for sudo in terminal. However, if I let my laptop sleep/hibernate or manually lock it, then Howdy is not working for that login screen that is part of Manjaro KDE. I cannot figure out what I need to change, or what that login screen would be called since it is not GDM after initial login. I have added 'auth sufficient pam_python.so /lib/security/howdy/pam.py' to the following PAM files trying to get it to work:

-     sudo
-     system-local-login
-     KDE (thought this would fix it)
-     login (might be unnecessary)
-     system-login (might be unnecessary)

Thanks in advance for any help!

boltgolt, 2020-09-28:
You might have to try to set more broader permissions for the files in `/lib/security/howdy`, you can find the exact `chmod` command in a few other threads

mortie, 2020-11-20:
I encountered a similar issue to this just now, using the official Ubuntu PPA. I was unable to use howdy with swaylock - `ModuleNotFoundError: No module named 'recorders.video_capture'` - because swaylock doesnt run as root.

Is there no way for the Howdy packages to set the executable bits to the correct directories? It would've been nice if howdy worked with non-suid programs out of the box.

X-Ryl669, 2020-11-20:
@mortie Do you know what user swaylock run as ? If so, you'll need to add this user to a group (for example biologin) and also set howdy files to that group (and /dev/videoXXX). Then you don't need to run as root anymore.

BackMountainDevil, 2021-04-29:
# version

- Howdy 2.6.1
- Linux arch 5.11.6-arch1-1
- KDE 5.21.2

## pam config

- /etc/pam.d/sudo
- /etc/pam.d/system-local-login
- /etc/pam.d/sddm

described in my [bolg](https://backmountaindevil.github.io/post/linuxhello/)

# good

- sudo
- start login

# bad

same as u

rhymeswithmogul, 2022-02-02:
I'm having an almost-identical issue on Ubuntu 21.10 and Howdy 2.6.1.  I can run all of the `howdy` commands, but it fails to work as a PAM module.

I did install the OpenCV Python module (Ubuntu: `apt install python3-opencv`).

```
user@computer:~$ sudo -s
Attempting face detection
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture
ModuleNotFoundError: No module named 'recorders.video_capture'
Unknown error: 1
```

I've tried running `chmod -R 755 /lib/security/howdy` like the wiki mentions, but there was no change afterwards.

closed 2026-11-16 by boltgolt