boltgolt/howdy #449: GUI for Howdy (config especially)

LorenDB, 2020-09-24:
I am interested in creating a Qt-based GUI for configuration of Howdy for Hacktoberfest. Would this be approved?

LorenDB, 2020-09-25:
> I am interested in creating a Qt-based GUI for configuration of Howdy for Hacktoberfest. Would this be approved? 

    I think that I will create an organization for hosting this project myself.

closed 2020-09-25 by LorenDB

boltgolt, 2020-10-03:
Hey! Sorry for not getting back to you sooner, i don't monitor closed issues very well. A GUI would be very much appreciated, especially if it's in something like GTK so it could be distributed as `howdy-gkt` for instance

LorenDB, 2020-10-03:
OK, we'll see what I come up with. I don't know GTK, so it will need to be Qt.

mcgivrer, 2021-12-03:
Any news on this very interesting subject ?

LorenDB, 2021-12-03:
Well. It turns out that I gave up on the idea since my laptop's camera isn't IR and therefore is very bad for facial recognition. Maybe someday if and when I get an IR camera....

boltgolt, 2021-12-03:
@mcgivrer there's a working GUI on the beta branch, but it's waiting for other changes in the 3.0.0 version

SaadBazaz, 2022-12-28:
Really would be great, so the UX is better when adding faces. It'd be great if it could have a shortcut in Settings.