boltgolt/howdy #450: Missing module recorders.video_capture

ng2303, 2020-10-01:
Whenever howdy tries to run I get this error. Test is working btw. 

traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture
ModuleNotFoundError: No module named 'recorders.video_capture'
Unknown error: 1

I searched and didn't find any similar issues. What should I do? After this happens when I try to use sudo I'm prompted to just enter my password.

ng2303, 2020-10-01:
I'm running Pop OS on a Dell XPS 13 if that helps.

ng2303, 2020-10-01:
It stopped doing that and is now giving me this error - 
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Face detection timeout reached

I am running 2.6.1, the latest release, as well, since similar documentations of this issue being fixed were for previous versions.

Zocker1999NET, 2020-10-02:
Running Debian Unstable, having the same issue with howdy 2.6.1 and 2.6.0, had not with 2.5.1. I think this might a dependency issue, tried to install `python3-opengl` manually but does not help

petre2dor, 2020-11-01:
Running Ubuntu 20.10 with KDE Plasma. Lock screen fails, but sudo and other prompts work.

I managed to debug the issue by running:
`/usr/lib/x86_64-linux-gnu/libexec/kscreenlocker_greet --testing`

The output is:

```
OpenGL vendor string:                   NVIDIA Corporation
OpenGL renderer string:                 GeForce GTX 1650/PCIe/SSE2
OpenGL version string:                  4.6.0 NVIDIA 455.28
OpenGL shading language version string: 4.60 NVIDIA
Driver:                                 NVIDIA
Driver version:                         455.28
GPU class:                              Unknown
OpenGL version:                         4.6
GLSL version:                           4.60
Linux kernel version:                   5.8
Requires strict binding:                no
GLSL shaders:                           yes
Texture NPOT support:                   yes
Virtual Machine:                        no
kf.package: No metadata file in the package, expected it at: "/home/xxxx/.local/share/wallpapers/"
file:///usr/share/plasma/wallpapers/org.kde.image/contents/ui/main.qml:76:9: Unable to assign [undefined] to QStringList
file:///usr/share/plasma/wallpapers/org.kde.image/contents/ui/main.qml:75:9: Unable to assign [undefined] to int
Locked at 1604226833
UdevQt: unable to create udev monitor connection
Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/SessionManagementScreen.qml:64:5: Unable to assign [undefined] to int
file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/ActionButton.qml:33:5: Unable to assign [undefined] to int
file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/KeyboardLayoutButton.qml:29:5: Unable to assign [undefined] to int
file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/Battery.qml:48:9: Unable to assign [undefined] to double
file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/UserDelegate.qml:41:5: Unable to assign [undefined] to int
qt.svg: <input>:406:376: Could not add child element to parent element because the types are incorrect.
kf.package: No metadata file in the package, expected it at: "/home/xxxx/.local/share/wallpapers/"

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture
ModuleNotFoundError: No module named 'recorders.video_capture'
Authentication failure
```
Hope this helps in the debug process.

bartp1990, 2020-11-02:
I experienced a similar issue on my Dell XPS 9500 running popOS 20.10. 
After digging around I noticed that folder `/lib/security/howdy/recorders` did not have the right permissions. 
Changing them with `sudo chmod a+x /lib/security/howdy/recorders` solved the problem for me.

petre2dor, 2020-11-17:
:+1: solved by @bartp1990 's solution. 
thank you so much :)

Zocker1999NET, 2020-11-18:
I got on Debian Unstable after re-upgrading from howdy 2.5.1 to 2.6.1 another issue:
```
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 283, in <module>
    make_snapshot("SUCCESSFUL")
  File "/lib/security/howdy/compare.py", line 55, in make_snapshot
    snapshot.generate(snapframes, [
  File "/usr/lib/security/howdy/snapshot.py", line 54, in generate
    os.makedirs(abpath + "/snapshots")
  File "/usr/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/usr/lib/security/howdy/snapshots'
```

Said directory does not exist and permissions of parent would only allow root to create a directory (stripped other dirs/files):
```
$ ls -la /usr/lib/security/howdy
drwxr-xr-x 8 root root 4096 18. Nov 18:04 .
```

after creating the directory manually (`sudo mkdir /usr/lib/security/howdy/snapshots`) everything works for me too. Thanks to @bartp1990 

**Note:** `/lib` is symlinked to `/usr/lib` on my system.

closed 2026-11-16 by boltgolt