boltgolt/howdy #451: Add support for biometric encryption

JeWe37, 2020-10-10:
As it stands it is not possible to use howdy to open say kwallet, as howdy cannot unlock keys that are stored in a cryptographically secure manner. However, techniques that attempt to address this do exist under the name biometric encryption and could potentially(with the proper adjustments to the pam.py script) allow that such vaults could be unlocked with howdy. This could also allow for instance encrypted partitions to be unlocked using biometrics alone.

I might investigate this issue myself at some point, though I am somewhat short on time at the moment.

Zocker1999NET, 2021-04-12:
This might be an interesting idea but I think this is theoretically not possible in a really secure way.

Currently, I see the problem that howdy will not be able to derive a key of your face, but for some applications (like a password vault per user on a trustworthy secure multi-user system), it might be sufficient that howdy stores a private key in plain text somewhere on your drive protected by root privileges. This key can then be used by howdy to unlock e.g. your vault after a successful authorization. But it would not protect your key from being extracted from your computer while having physical access or other applications/users with root privileges.

That's mostly because not the camera itself but howdy does the face detection. For example on newer Android phones, the fingerprint sensor is itself a token device capable of storing tokens. Android can only use those tokens (but not really read them) to derive the required passphrases while the user places a known finger on the reader because not Android does the detection and knows your finger but the reader itself can do this. Android only instructs the fingerprint reader to e.g. add a new finger, delete older ones, verify the next finger, …

boltgolt, 2021-04-13:
Difficult to do, but any practical ideas on how to implement this in Howdy specifically are very welcome