boltgolt/howdy #453: Add support for Clear Linux

rayzorben, 2020-10-16:
Just looking for an install script that could apply across any distro. If not possible, just some guidance for Clear Linux.

I tried getting it working, and used the Fedora installer as a guide.

1. moved src/* to /lib/security/howdy
2. ran dlib-data install script and moved .dat files to /lib/security/howdy/dlib-data
3. ran `install -Dm 0644 fedora/com.github.boltgolt.howdy.policy' to /usr/share/polkit-1/actions
4. skipped bash completion for now
5. linked /usr/bin/howdy to /lib/security/howdy/cli.py
6. installed missing packages with pip: `numpy, opencv-python, dlib`
7. installed qt-basic with swupd
8. had issues with scripts `import ConfigParser` is now `import configparser`
9. `compare.py, pam.py, snapshot.py` were not executable, so I made them `chmod +x`
10. above scripts still wouldn't run so I added `#!/usr/bin/env python3` to them

Still wasn't working, so I tried adding the `/etc/pam.d` files as mentioned in the Arch Linux wiki. For example, `sudo`

```
auth    sufficient      pam_python.so /lib/security/howdy/pam.py
auth    include         system-auth
account include         system-auth
session include         system-auth
```

When I try to run `sudo su -` it just prompts me for my password, no face recognition.

If I remove the bottom 3 lines, I get an error `sudo: account validation failure, is your account locked?`

boltgolt, 2020-10-21:
DEFINITELY dot not remove the other 3 lines for sudo, this could break your whole installation if you're not careful. I guess your issue is with the python 3.8 installation, have you tried reinstalling that or switching to python 3.6?

rayzorben, 2020-10-21:
@boltgolt python 3.8 is what it comes with, unfortunately I can't test 3.6 without significant effort

> Clear Linux versions bundles for the whole operating system. It is not possible to downgrade individual packages. You would have to either build the package yourself, rollback the whole operating system to an older version, or wait for a fix from our team

It would be easier to fix it to work with 3.8, but i'm not even sure where to start. Are there some basic tests I can do to isolate the issue?

rayzorben, 2020-10-22:
So I got face recognition working for sudo, no luck yet with gdm.

Seems pam-python is embedding python3.8 and trying to embed libpython3.8.a. When I switch that to libpython3.8.so it works. Not sure if that is the right thing to do.

Paritosh97, 2020-11-11:
> So I got face recognition working for sudo, no luck yet with gdm.
> 
> Seems pam-python is embedding python3.8 and trying to embed libpython3.8.a. When I switch that to libpython3.8.so it works. Not sure if that is the right thing to do.

Encountered a similar situation. You can see my 'pam.d/sudo' file below. However, I can't make it work for sudo.
`#%PAM-1.0
auth sufficient libpython3.8.so /lib/security/howdy/pam.py
auth		include		system-auth
account		include		system-auth
session		include		system-auth
`