boltgolt/howdy #454: Fedora no module named 'dlib'

dooman17, 2020-10-19:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Attempting to run sudo howdy add or sudo howdy test gives an error for dlib. I am on a surface laptop 3 running the surface linux kernel

```
$ sudo howdy test
Traceback (most recent call last):
  File "/usr/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib64/security/howdy/cli/test.py", line 9, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
```

```
$ sudo howdy add
No module named 'dlib'

Can't import the dlib module, check the output of
pip3 show dlib
```

----

https://github.com/boltgolt/howdy/issues/124 says this was an issue at one point but it was fixed

Linux distribution: fedora

Howdy version 2.6.0

principis, 2020-10-20:
@dooman17 which fedora version?

dooman17, 2020-10-20:
> @dooman17 which fedora version?

32

principis, 2020-10-20:
> > @dooman17 which fedora version?
> 
> 32

I think you're using luya's copr repo? Try mine over [https://copr.fedorainfracloud.org/coprs/principis/howdy/](here).

justitsi, 2020-10-28:
Hi, I just managed to install it: `sudo dnf install dlib python-dlib `

JediBrooker, 2020-10-31:
> Hi @dooman17, I just managed to install it: `sudo dnf install dlib python-dlib ` (works on fedora 33, you need to fiddle with PAM to get it working for sudo)

i've got it working for sudo on fedora 33 but i can't get it to work when logging in. i get the "unknown device 1" error. i've tried the other fixes mentioned such as changing no_confirmation to true, as well as changing the permissions of /lib64/security/howdy but still not working.

i'm new to fedora so i'm struggling to troubleshoot here. i can't even find /var/log/auth.log because it doesn't exist so i tried journalctl SYSLOG_FACILITY=10 but i can't identify anything obvious there. what should I be looking for?

principis, 2020-11-06:
@JediBrooker Try disabling selinux, also the log file is in `/var/log/secure`

I've seen people mentioning that too much output breaks their login screen, see #381.  I've fixed that in a different copr repo.
You can find it over [here](https://copr.fedorainfracloud.org/coprs/principis/howdy-patched/). Remove the other repo and howdy before installing this version!

closed 2026-11-16 by boltgolt