boltgolt/howdy #456: Skip facial detection after password entered

ralphocdol, 2020-10-24:
There are times when my face isn't detected properly due to poor lighting environment.

In `pkexec` case or the GUI authentication, the facial recognition waits for the face to be detected or timeout to reach even after the password was already entered.

In `sudo` in terminal/console, the facial detection is asked first before the password (when it fails).

It would be better if the password is asked while the facial recognition is detecting the face then:
- Skip the face detection if password is entered
- Skip password when face is detected

I don't know if this can be achieved but I think this could help us do things seamlessly.

Thanks :)

FedeCuci, 2020-10-28:
+ 1 for me. Came here to open the exact same thing.

X-Ryl669, 2020-11-20:
As far as I've understood, it's not possible. This is not a limitation in Howdy but in PAM as it's not multithreaded, so it tries one module after the other in order. There used to be hack by allowing, in one module, multiple input (like stdin (keyboard) input + other method), but it breaks the security of PAM where each module is audited for what it does.

So, you should report a bug to PAM developers to allow multi-thread/multi-process authentication. The same issue arise with fprint for fingerprint authentication.

FedeCuci, 2020-11-21:
Is this their github repo: https://github.com/linux-pam/linux-pam? @X-Ryl669

X-Ryl669, 2020-11-21:
Yes, but for an example implementation of a "parallel" implementation, see [here](https://unix.stackexchange.com/questions/207360/use-multiple-pam-login-methods-at-once). It requires hacking the pam module to implement stdin/password capture and, honestly, it looks like it's a hack than a genuine operation.

FedeCuci, 2020-11-21:
I'd be happy to open an issue on their GitHub, I just wouldn't really know how to explain myself as I don't know how the PAM system works apart from you said above @X-Ryl669.

X-Ryl669, 2020-11-21:
I've started an issue on their repo. Cheers!

FedeCuci, 2020-11-21:
That's nice of you, thanks! @X-Ryl669