boltgolt/howdy #458: ModuleNotFoundError: No module named 'ConfigParser'

HughPH, 2020-10-28:
Python never fails to give me a headache.

Howdy just wasn't even starting the webcam, so I ran `python3 /lib/security/howdy/pam.py` and I got `ModuleNotFoundError: No module named 'ConfigParser'`

So I installed it:
```bash
$ sudo pip3 install ConfigParser
Collecting ConfigParser
  Downloading configparser-5.0.1-py3-none-any.whl (22 kB)
Installing collected packages: ConfigParser
Successfully installed ConfigParser-5.0.1
WARNING: You are using pip version 20.2.1; however, version 20.2.4 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
```

Then ran pam.py again:
```bash
$ sudo python3 /lib/security/howdy/pam.py 
Traceback (most recent call last):
  File "/lib/security/howdy/pam.py", line 10, in <module>
    import ConfigParser
ModuleNotFoundError: No module named 'ConfigParser'
```

What gives?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04

Howdy version (`sudo howdy version`): 2.6.1

arifer612, 2020-11-09:
You need to have python 2 for it to work. This is because the `pam-python` module only operates on python 2. If you have python 2 you should not have to worry about `ConfigParser` not being installed because it should be part of the standard library.

boltgolt, 2020-11-23:
@arifer612 is exactly right! ConfigParser is not the issue here, did you notice any errors in `/var/log/auth.log`?

closed 2021-06-14 by boltgolt

pavelkq, 2022-11-28:
My Debian 11 uses python3 as a default despite of I set up python2. How could I force to use python2 for howdi?