boltgolt/howdy #460: Ubuntu server 20 LTS - Howdy add fails. 'VideoCapture' object has no attribute 'internal'

kanaida2, 2020-10-31:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

kanaida@kanaida-laptop:~/Downloads$ sudo howdy add
Adding face model for the user kanaida
Enter a label for this new model [Initial model] (max 24 characters): kanaida
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f7cbc4edee0>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

kanaida@kanaida-laptop:~/Downloads$ sudo howdy snapshot
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f1cd1abaf70>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

kanaida2, 2020-10-31:
Missing config path was the problem.
It should probably say that when i try to add, or try to set the default to /dev/video0 if one isn't provided by the distro.

sudo howdy config
changed device path to /dev/video0

boltgolt, 2020-11-01:
The error you've posted _does_ say:
```
 Camera path is not configured correctly, please edit the 'device_path' config value.
```

The ubuntu installer configures the device path automatically on installation. I'm guessing you're not using Ubuntu yourself but you haven't filled in your distro in the issue.

closed 2020-11-01 by boltgolt

wout, 2020-12-24:
I had this issue as well with elementary OS 5.1.7. Configuring the device path manually (`/dev/video0`) fixed it.