boltgolt/howdy #462: Cannot be built on Arch Linux

bobzhu418, 2020-11-02:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Because Arch Linux has removed Python2 packages from its repository, the latest Howdy cannot be built any more. Below is the error message:

```
 -> Could not find all required packages:
    python2-sphinx (Wanted by: howdy -> pam-python)
    python2-sphinx (Wanted by: howdy)
```
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux (Kernel 5.9.2)

Howdy version (`sudo howdy version`): 2.6.1

bageljrkhanofemus, 2020-11-06:
its because the pam-python package needs python2-sphinx to produce documentation. You can remove the documentation from the pkgbuild and the sphinx requirement from both pkgbuilds.

bobzhu418, 2020-11-07:
Yes. You're right. I've built the the package successfully. Thanks!

closed 2020-11-07 by bobzhu418

ginhiki, 2020-11-09:
> Yes. You're right. I've built the the package successfully. Thanks!

Facing the same issue, can you say how you rebuilt the package?