boltgolt/howdy #466: module video_capture not found

roycdavies, 2020-11-06:
I'm running PopOS 20.10.  Howdy installed fine, but when I attempt to use it, the following error arises:

roycdavies@starlabtop:~$ sudo ls
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture
ModuleNotFoundError: No module named 'recorders.video_capture'
Unknown error: 1
[sudo] password for roycdavies:

RedCoder09, 2020-12-24:
Try running "sudo howdy config" and set "device_path" to "/dev/video0".

tomasdisk, 2021-04-04:
yes, look for one of your camera devices with `ls -l /dev/video*`, you will get something like this:
```
crw-rw----+ 1 root video 81, 4 abr  3 22:33 /dev/video4
crw-rw----+ 1 root video 81, 5 abr  3 22:33 /dev/video5
crw-rw----+ 1 root video 81, 6 abr  3 22:33 /dev/video6
crw-rw----+ 1 root video 81, 7 abr  3 22:33 /dev/video7
```

and then do as @RedCoder09 said, open config file with `sudo howdy config`  and edit the line for `device_path`:
```
# before
device_path = /dev//dev/v4l/by-path/none
# after
device_path = /dev/video4
```

closed 2026-11-16 by boltgolt