boltgolt/howdy #467: Extremely Slow Launch

mosmash, 2020-11-07:
Hi, I've managed to get Howdy setup and running on KDE on Manjaro but it seems to be running terribly slow, especially during the camera launch. For reference these are  some typical post authentication results:
```Time spent
  Starting up: 184ms
  Open cam + load libs: 1373ms
    Opening the camera: 1373ms
    Importing recognition libs: 863ms
  Searching for known face: 510ms
  Total time: 2067ms

Time spent
  Starting up: 190ms
  Open cam + load libs: 1362ms
    Opening the camera: 1362ms
    Importing recognition libs: 836ms
  Searching for known face: 515ms
  Total time: 2067ms
```
Anyone know what could be causing this? I really want this to run as fast as Windows Hello does on this laptop, and I havent seen camera launch speeds this high before so I'm really not sure what's causing it. My laptop is also quite high performance so I'm really left bamboozled.

I had a search online for previous issues with speed and I found [#99](https://github.com/boltgolt/howdy/issues/99) but all the camera launch speeds there are nowhere near what I'm experiencing. 

----

System Specs : AMD Ryzen 7 4800u CPU / 16gb DDR4-3200 ram 

Distro: KDE Plasma on Manjaro 5.8.18.1 

Howdy Version: 2.6.1

daiaji, 2020-12-30:
```
Time spent
  Starting up: 102ms
  Open cam + load libs: 596ms
    Opening the camera: 596ms
    Importing recognition libs: 290ms
  Searching for known face: 316ms
  Total time: 1014ms

Resolution
  Native: 340x-1
  Used: 320x320

Frames searched: 2 (6.33 fps)
Black frames ignored: 0 
Dark frames ignored: 1 
Certainty of winning frame: 2.991
Winning model: 0 ("me")
```
Similar, but mine will be faster
Maybe I used a faster SSD
I guess putting Lib in memory might make it all faster

efschu, 2022-09-06:
Time spent
  Starting up: 123ms
  Open cam + load libs: 2856ms
    Opening the camera: 2856ms
    Importing recognition libs: 246ms
  Searching for known face: 123ms
  Total time: 3102ms

Resolution
  Native: 480x-1
  Used: 320x427

Frames searched: 1 (8.14 fps)
Black frames ignored: 0 
Dark frames ignored: 0 
Certainty of winning frame: 2.632

Debian 11 with gnome

Opening camera could be improved

closed 2026-11-16 by boltgolt