boltgolt/howdy #471: Arch linux pkgbuild outdated

Giannie, 2020-11-12:
The aur pkgbuild files for howdy and pam-python no longer work. They depend on python-sphinx2 which is no longer available. In addition, pam-python currently depends on python2, although my testing indicates there aren't any issues with depending on python3.

I would be willing to take on maintenance of these packages if it helps.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux

Howdy version (`sudo howdy version`): 2.6.1

Ranguna, 2020-11-13:
There's a fix in the aur package page by user bageljr, `python2-sphinx` can be replaced with `python-sphinx`: https://aur.archlinux.org/packages/howdy/

But the pam-python package also needs to be updated.

Giannie, 2020-11-14:
PR  #472 has updated pkgbuild files that should work.

Kombustor, 2020-11-14:
> PR #472 has updated pkgbuild files that should work.

Awesome, can you push it into the AUR?

Giannie, 2020-11-14:
I'm not a maintainer of the AUR packages so I don't believe I can push.

Kombustor, 2020-11-14:
> I'm not a maintainer of the AUR packages so I don't believe I can push.

@boltgolt @kageurufu

jinnko, 2020-12-02:
Big update came through on ArchLinux today and howdy was a casualty due to the broken dependencies.

Would really like to see #472 merged so multiple people don't end up having to manually shoehorn a solution.

Kombustor, 2020-12-02:
> Big update came through on ArchLinux today and howdy was a casualty due to the broken dependencies.
> 
> Would really like to see #472 merged so multiple people don't end up having to manually shoehorn a solution.
> 
> FWIW, I've also successfully used the changes in #472 to get Howdy working again.

How did you use the branch? Just manually cloned and installed howdy from there?

jinnko, 2020-12-03:
Yes, basically build from the PR branch.  It roughly looks like this:

```
sudo cp /lib/security/howdy/config.ini /lib/security/howdy/config.ini.backup
yay --remove --cascade howdy pam-python python2-sphinx
git clone https://github.com/Giannie/howdy.git
cd howdy/archlinux/pam-python
makepkg -si
cd ../howdy
makepkg -si
sudo cp /lib/security/howdy/config.ini.backup /lib/security/howdy/config.ini
```

During `makepkg -si` there may be unmet dependencies which you may have to resolve manually.

You may also find you need to remove other dependencies that may not have been cleaned up properly before everything falls into place.

kageurufu, 2020-12-03:
I updated the AUR to use python-sphinx and include the license for pam-python. I removed python-sphinx from the howdy PKGBUILD altogether.

@Giannie thanks for the PR, but it has an outdated version of howdy in the pkgbuild. 
I'll open a new PR with the current versions from AUR

closed 2020-12-07 by boltgolt