boltgolt/howdy #474: ctrl+z keeps camera busy

amirkaveh, 2020-11-18:
When howdy is recognizing a face while entering a sudo command, if someone press ctrl+z, the process will be sent to the background, while the camera IR emitter keeps working and it does not stop. I think it should release resources (camera) when it is sent to the background.

Linux distribution (if applicable): Ubuntu 20.10

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2020-11-23:
I'm confused as to why you would ever want to halt Howdy to be resuming it later? Wouldn't killing it and starting the authorization again be better?

amirkaveh, 2020-11-23:
Absolutely, It is not reasonable to halt such processes. But I think it is better to handle unintentional or even improbable situations like this.