boltgolt/howdy #475: Failfast if camera is in use

tdaniely, 2020-11-24:
I commonly encounter a situation where I'm in a google meet session, I'm running `sudo something` in console, or getting some other polkit prompt.
In this situation howdy fails because the camera is locked, But the recognition timeout is applied so I have to wait the timeout until I can enter my password.

Can quick fix this by either failing fast if the camera is locked, or having a seperate timeout config for when device is in use.

boltgolt, 2020-12-03:
That's weird, when the camera is busy it should throw an error that it can't open the device. Are google meet and Howdy using the same video device?

tdaniely, 2020-12-03:
Actually no, I symlinked the device as I described in the multiple camera issue, I'll test it without the symlink and post back.

Doomsdayrs, 2021-06-20:
@tdaniely Results?

tdaniely, 2021-06-20:
Sorry I completely forgot about this.
It was due to a hack I did to switch to the dock camera when the laptop is docked.

closed 2021-06-20 by tdaniely