boltgolt/howdy #476: Run Howdy and fprint in parallel?

Awbmilne, 2020-12-01:
_First off, Not a feature request. Just a question targeted at people who might have a good answer._

I am switching to a new XPS laptop and have been looking at using proper 2FA for login. I would like to be able to login with 2 of 3 factors:
- Something I Know: _Pin_
- Something I Have: _Yubikey with HMAC Challenge Response_
- Something I Am: _Howdy IR or fprint Fingerprint_

The first two are easy to deal with through `pam_unix` and `pam_yubico`.\
What I would like to be able to do is:
1. Use Pin & Yubikey | Enter Pin, HMAC succeeds, _Sufficient_
2. Use Pin & Howdy | Enter Pin, HMAC Fails & passes, Howdy succeeds, _Sufficient_
3. Use Pin & Fingerprint | Enter Pin, HMAC Fails & passes, fprint succeeds, _Sufficient_
4. Use Yubikey & Howdy | Enter blank pin, HMAC Succeeds, Howdy succeeds, _Sufficient_
5. Use Yubikey & Fingerprint | Enter blank pin, HMAC Succeeds, fprint succeeds, _Sufficient_

The Pin and HMAC can be easily handled linearly using `pam_unix` and `pam_yubico`, but... \
***How could Howdy and fprint be handled in parrallel so that you dont need to wait for one to fail to use the other?***

Is there a way to spawn multiple PAM sessions for the same authentication? Is there a way to do this by creating a new parrallel Pam module?

As an Extra thought experiment, Could all 4 PAM modules be run in parrallel with the correct combination allowing Auth (<- Probably not).

Awbmilne, 2020-12-01:
It seems my search after posting this was better than my search before.

#9 Is very similair, but slightly different because passwords can be stored and regurgitated.

What I stated above is likely require PAM style success/fail handling in parrallel, Which boltgolt has mentioned doesnt really work since PAM blocks threads. If anyone Stil

boltgolt, 2020-12-03:
It's not impossible but if there's no premade PAM module it's going to be very difficult. The good news is that PAM only blocks the python interpreter, so if you manage to launch a separate process for the fingerprint module it could theoretically work

Awbmilne, 2020-12-07:
Would either Multithreading or Multiprocessing allow python to handle Multiple tasks without the PAM blocking the whole thing?

For example, Could a PAM module written in Python call 3 other PAM modules and return authentication based on if at least 2 of the factors returned true?

There would obviously need to be some weird way to configure this Intermediate Parallel-PAM module, But from your experience, would this possibly work?

boltgolt, 2020-12-07:
It could possibly work but don't underestimate the effort involved in that. You'd have to emulate the PAM interface in C somehow, which would a real PIA

Awbmilne, 2020-12-07:
I believe there are already tools for this in Python, Namely the [`pam` library](https://atlee.ca/software/pam/index.html) for Python. 

Looking at the [documentation](https://atlee.ca/software/pam/module-pam.html#functions) and the [source code](https://atlee.ca/software/pam/pam.py.html) for this library, It seems to just wrap PAMLIB C code in a Python `authenticate()` function. This `authenticate()` function takes `username`, `password`, and `service` arguements. The service defaulting the standard PAM login service.

With that, it seems a set of PAM services could be setup, Each comprising of just 1 factor of Authentication (Howdy, fprint, YubikeyHMAC, etc.) set as `sufficient` could be created, with threads of the Primary Python PAM handler calling one instance of each simultaneously for the same user. It could then handle the outputs and decide if it was overall sufficient or not.

I am unsure about how PAM would recognize a call to `authenticate()` in the `pam` library.  Specifically if it recognizes threads/processes on python as seperate-enough to have seperate PAM sessions. _Also_, If this did work, it would be a little klugey with 3 (or more) PAM services and a Parrallel PAM python tool.

closed 2023-03-28 by boltgolt