boltgolt/howdy #479: howdy add error No module named 'numpy'

cloudsun9, 2020-12-17:
$ sudo howdy add 
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 10, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

then
$ sudo howdy snapshot
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 107, in <module>
    import cli.snap
  File "/usr/lib/security/howdy/cli/snap.py", line 7, in <module>
    import snapshot
ModuleNotFoundError: No module named 'cv2'

  File "/usr/lib/security/howdy/snapshot.py", line 4, in <module>
    import cv2

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution Ubuntu 20.10

Howdy 2.6.1

skewballfox, 2020-12-20:
check if python-numpy and python-opencv are currently installed on your system

cloudsun9, 2020-12-20:
I don't know a good way to check, so for numpy I installed numpy-dice-hy. then tried howdy add.
Same result. 
Could the path be different in 20.10?

c

cloudsun9, 2020-12-20:
[image]
[image]

cloudsun9, 2020-12-21:
$ pip3 show numpy
Name: numpy
Version: 1.19.4
Summary: NumPy is the fundamental package for array computing with Python.
Home-page: https://www.numpy.org
Author: Travis E. Oliphant et al.
Author-email: None
License: BSD
Location: /home/david/.local/lib/python3.8/site-packages
Requires: 
Required-by: scipy, tifffile, scikit-image, PyWavelets, opencv-python, matplotlib, imageio, wxPython

cloudsun9, 2020-12-21:
pip3 install opencv-python
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: opencv-python in /usr/local/lib/python3.8/dist-packages (4.4.0.42)
Requirement already satisfied: numpy>=1.17.3 in /home/david/.local/lib/python3.8/site-packages (from opencv-python) (1.19.4)

erratic-c-programmer, 2020-12-27:
Try `sudo pip3 install opencv-python numpy` to install the packages in system directories

cloudsun9, 2020-12-27:
$ sudo pip3 install opencv-python numpy
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 17, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
Unknown error: 1
[sudo] password for david: 
Requirement already satisfied: opencv-python in /usr/local/lib/python3.8/dist-packages (4.4.0.42)
Requirement already satisfied: numpy in /usr/local/lib/python3.8/dist-packages (1.19.4)

cloudsun9, 2020-12-27:
Then...

sudo howdy add
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 10, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

boltgolt, 2020-12-27:
Extremely weird, what you did should have fixed the issue. Have you tried reinstalling Howdy itself?

cloudsun9, 2020-12-27:
Uninstalled and reinstalled multiple times. 
I wonder if it is to with the fact I am doing this on Ubuntu 20.10.
I asked Slimbook for help with Slimbook Face, before I found out it was just a GUI for Howdy, and they said it works on 20.04 but that they have no ideas about 20.10.

VanVan, 2021-05-15:
@boltgolt Everything worked fine for me on Ubuntu 20.10, but this problem came after updating to Ubuntu 21.04

cloudsun9, 2021-05-15:
david@pixie:~$ sudo pip3 install numpy
[sudo] password for david: 
Requirement already satisfied: numpy in /usr/local/lib/python3.9/dist-packages (1.20.2)
WARNING: Running pip as root will break packages and permissions. You should install packages reliably by using venv: https://pip.pypa.io/warnings/venv
david@pixie:~$ sudo pip3 install opencv-python
Requirement already satisfied: opencv-python in /usr/local/lib/python3.9/dist-packages (4.5.1.48)
Requirement already satisfied: numpy>=1.19.3 in /usr/local/lib/python3.9/dist-packages (from opencv-python) (1.20.2)
WARNING: Running pip as root will break packages and permissions. You should install packages reliably by using venv: https://pip.pypa.io/warnings/venv
david@pixie:~$ sudo pip3 install dlib
Collecting dlib
  Downloading dlib-19.22.0.tar.gz (7.4 MB)
     |████████████████████████████████| 7.4 MB 2.5 MB/s 
Building wheels for collected packages: dlib
  Building wheel for dlib (setup.py) ... done
  Created wheel for dlib: filename=dlib-19.22.0-cp39-cp39-linux_x86_64.whl size=4353319 sha256=9b48f6de28b42af522e3c249c25d2d89cca26ec18d7c78f095a7a00a640df4fc
  Stored in directory: /root/.cache/pip/wheels/33/2c/16/6ceb1bda8e67571304aaf3cd80c8cb47dc8d5ab99e34bda88b
Successfully built dlib
Installing collected packages: dlib
Successfully installed dlib-19.22.0
WARNING: Running pip as root will break packages and permissions. You should install packages reliably by using venv: https://pip.pypa.io/warnings/venv
david@pixie:~$

cloudsun9, 2021-05-15:
Ubuntu 21.04
This is new.
Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 110, in <module>
    handleStatus(sc(["pip3", "install", "--upgrade", "pip"]))
  File "/usr/lib/python3.9/subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3'
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit
 status 1
Processing triggers for man-db (2.9.4-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
david@pixie:~$

VanVan, 2021-05-15:
The problem was fixed for me by installing only the packages specified in my previous message.
I didn't invest more, but if you don't use python packages for other programs, you can uninstall all the python packages and howdy and try to reinstall the whole thing, or even take the opportunity to upgrade to the next version 21.04 of ubuntu

NightMasterX, 2021-11-30:
This is still a problem...

EmixamPP, 2021-12-04:
Maybe using the distribution's repo is a better idea than using pip: `sudo apt install python3-numpy`?

cloudsun9, 2021-12-04:
sudo apt install python3-numpy 
Done long ago. I get the same with other python solutions though, so not your fault.

EmixamPP, 2021-12-04:
Wow, I didn't see that this issue was almost a year old.
Reinstalling Ubuntu (or better distro, yeah I don't like Ubuntu) would be a solution to consider :/

Lyton505, 2022-09-06:
Same issue on Pop_OS 22.04

Lyton505, 2023-03-06:
Hey @boltgolt are you still looking into this issue? I can confirm this is a problem on PopOS 22.04 too

boltgolt, 2023-03-09:
It usually seems to be an issue where the python environment pip uses is not the one that Howdy will load into. Maybe:
```
python3 -m pip install numpy
python3 -m pip install opencv-python
python3 -m pip install dlib
```

cloudsun9, 2023-03-09:
Nope, that's still not it.
I believe howdy is looking in the wrong (old?) locations.
I'm on Ubuntu 22.10 now.

boltgolt, 2023-03-10:
Howdy just imports like any other python3 script. Maybe you can look into similar issues with other python programs?

esseswann, 2023-08-07:
I think I can confirm that this happens after release upgrade. I upgrdaded from 22.04 to 23.04 and this happened

cloudsun9, 2023-08-07:
Disappointing.

musikid, 2023-12-23:
Do you have multiple versions of Python installed on your system?

rb58853, 2024-02-22:
this work for me on debian 12:
`$nano ~/.bashrc`
in the end of this file(bashrc) write: `export PATH="$HOME/.local/bin:$PATH"`
save the file with `ctrl+x`, `y`, `enter`
`source ~/.bashrc`
reinstall te packages (`numpy`, `opencv-python`,`dlib` )

keetha1011, 2024-05-26:
Linux Mint 23.1 has python 3.10 and default pip installs the package to python3.10 and not available to python3.11. 
Reinstalling the packages for python3.11 fixed for me.

yuyalun-allen, 2024-07-13:
> Do you have multiple versions of Python installed on your system?

The same problem, I think the reason is I use pyenv to install multiple versions of python along with python312 installed by pacman on arch linux. But how to solve this problem?

closed 2026-11-16 by boltgolt