boltgolt/howdy #482: Doesn't work in python 3.9

ArjunSahlot, 2020-12-23:
I have tried to install howdy in python 3.9 multiple times yielding unfortunate results.
/usr/local/lib/ after howdy installation:
[image]

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2020-12-31:
What kind of unfortunate results? Howdy should not touch `/usr/local/lib/`

ArjunSahlot, 2020-12-31:
There was always a dlib import error. I tried sudo pip3 install dlib but that would not resolve the issue.

wyapx, 2021-04-09:
> 
> 
> There was always a dlib import error. I tried sudo pip3 install dlib but that would not resolve the issue.

paste the traceback?

closed 2026-11-16 by boltgolt