boltgolt/howdy #483: Feature Request: Support confirmation before authorizing.

zeroby0, 2020-12-25:
Hi,

Thanks for this awesome tool!

At the moment, Howdy doesn't ask for confirmation before authorizing. This is perfectly fine when logging in, but may not be safe to do when authorizing `sudo`.

A script trying to elevate it's privileges could have authorization without the user's consent, and maybe without even the user's knowledge in case of background scripts. Windows Hello deals with this by showing a Modal asking for users approval. Fido keys do this by authorizing only when the user touches the button on them. Can we have a Modal asking for approval too?

I have not found a way to do this in the ReadMe / WiKi / Issues, apologies if this is already a feature. If it's not yet, please consider adding it.

Thanks!

zeroby0, 2020-12-25:
Perhaps we could even have the [head bobble](https://www.google.com/search?q=indian+head+bobble) as a gesture for approval instead of having to touch the computer  😂

boltgolt, 2020-12-31:
Thanks for the suggestion! This has been asked before and is currently being implemented on the dev branch

closed 2020-12-31 by boltgolt