boltgolt/howdy #486: HELP NEEDED: Translate Howdy to your language

boltgolt, 2020-12-31:
The next release of Howdy will have support for internationalization and translations to any language are welcome! If you want to help [join us on Crowdin](https://crwd.in/howdy). If your language is not listed yet leave a comment and i'll add it ASAP

Thanks in advance!

avelakafo, 2021-01-02:
I don't see Esperanto listed but I would be willing to undertake translating Howdy into it.

boltgolt, 2021-01-03:
I've added esperanto! @avelakafo

srueegger, 2021-01-17:
I started the German translation today and will finish it over the next week.
Good would be a German (Switzerland) translation option (which I would also directly translate).

boltgolt, 2021-01-17:
Thanks a lot @srueegger! I've added German (Swiss) to the list of languages

srueegger, 2021-01-22:
German Translation is complete.

Is it possible to copy the German translation 1: 1 to German (Swiss)? I just have to adjust a few strings.

boltgolt, 2021-01-22:
Thanks @srueegger! I've copied the german translations to swiss, it did not catch any duplicated translations but you should be able to easily copy those from the translation memory

srueegger, 2021-02-07:
Thanks for copying the language. German (Swiss) is also completely translated.

Toufupi, 2021-02-11:
Translation of Chinese simplified is finished! Thanks for your great work  creating such a useful tool! By the way when will you release the new verision of howdy with gui? I think it is fancy too!

boltgolt, 2021-02-11:
Thanks for your work @Toufupi! The GUI is going to be part of the larger v3.0 update which will break compatibility with older versions. Because of that i want to have as many breaking changes as possible in that one release and a couple of things are still being worked on. In short, no release date yet

agnibha92, 2021-03-02:
How to request other languages?

boltgolt, 2021-03-02:
@agnibha92 what language would you like me to add? There's unfortunately no way to let you add one yourself so just let me know what language you want

agnibha92, 2021-03-10:
I can translate to Hindi/Bengali, If you want to add these to the list... @boltgolt

boltgolt, 2021-03-10:
@agnibha92 I've added both languages to the list!

agnibha92, 2021-03-11:
Thanks @boltgolt

NedimBadzak, 2021-05-09:
I can translate to Bosnian/Croatian. @boltgolt

samuelgregorovic, 2021-06-24:
@boltgolt i can translate to Slovak language if you add it there

boltgolt, 2021-06-29:
@samuelgregorovic I've added Slovak!

samuelgregorovic, 2021-06-29:
> @samuelgregorovic I've added Slovak!
thanks! i will start ASAP and let you know when im done

boltgolt, 2021-06-29:
@samuelgregorovic no rush!

samuelgregorovic, 2021-07-01:
@boltgolt translations are 100% done :)

boltgolt, 2021-07-01:
@samuelgregorovic very quick, thank you!

samuelgregorovic, 2021-07-01:
np ✌️

liimee, 2021-07-06:
I'd love to translate it to Indonesian :)

boltgolt, 2021-07-07:
@liimee added!

simonpla, 2021-07-09:
Added a translation to Polish.

boltgolt, 2021-07-12:
@simonpla thank you!

simonpla, 2021-07-13:
@boltgolt No problem. :smile:

crispy245, 2021-08-03:
@boltgolt Finished the Spanish translation! Hope to support the actual codebase soon; the project seems really interesting!

boltgolt, 2021-08-05:
@crispy245 thanks a lot!

chamik, 2021-08-09:
Translated to Czech. Glad I could help!

boltgolt, 2021-08-10:
@chamik thank you!

Mertszr02, 2021-11-15:
@boltgolt I finished the Turkish translation but Crowdin shows 0 approved strings. I'm guessing approval is needed before the strings can be implemented. Is there anything I can do to help with that? 
btw thank you for creating Howdy it is incredible. I'm in the process of learning how Howdy works so I can help speed up the 3.0 release by Committing code.

boltgolt, 2021-11-15:
@Mertszr02 that just means that Crowdin is waiting on other Turkish speakers to verify your translation. I'm sure your translation is good so I'm going to import it regardless.

Good to hear! We especially need people to test the changes in #484

Itotutona, 2022-01-13:
I want to translate howdy in french

boltgolt, 2022-01-13:
@Itotutona I'd be very grateful if you could [verify the translations](https://translate.howdy.boltgolt.nl/project/howdy/fr) already done

boltgolt, 2022-04-14:
Thanks @gogoname! I'll take a look what i can do for LTR languages

NP-chaonay, 2022-05-24:
I want to request "Thai" language (language code: th)

I don't sure if I can able to do all 100% but I think most translation is not much difficult.

Thanks.

boltgolt, 2022-05-24:
Added @NP-chaonay, thanks!

GWFrank, 2022-06-01:
The rest of Traditional Chinese translation is done.
Would be great if someone can help verifying and do some polishing.

Glad that I can participate in this amazing project!

boltgolt, 2022-06-01:
@GWFrank thank you!

antival3nt, 2022-08-18:
Hey @boltgolt 
No software really used to be translated to Danish, I'd love to do it here, if it's okay? :)

saltedcoffii, 2022-09-18:
Hey @boltgolt could you divide Spanish into "Spanish, Spain", and "Spanish, Latin America"? You could also further subdivide latin american spanish into "Spanish, Mexico" and "Spanish, Argentina" if you wanted.

gimdh, 2022-10-19:
I have translated Korean, but there are couple of ambiguities that require confirmation. I intentionally left CLI script as English as I believe most Koreans willing to use CLI is comfortable with English. @boltgolt

boltgolt, 2022-10-31:
Sorry for the delayed response!

@antival3nt Of course! Danish should be ready to translate on Crowdin

@gimdh Thanks! Are the 55% of the strings that are not translated just the CLI?

gimdh, 2022-11-02:
@boltgolt Yes. _Main UI / Onboarding UI / UI scrpts_ are completely translated, though few clarifications would make it more clear.

drcor, 2022-11-17:
Hi @boltgolt 
I finished the missing translations for Portuguese (Portugal), now it is 100% done :)

Edyburr, 2022-12-08:
Hey @boltgolt !
I started the Romanian translation today and should be able to finish it by the end of the week.
Cheers!

BloodDrag0n, 2023-02-07:
Is Tamil translation done or you guys need any help?

kuracode, 2023-02-28:
Hi @boltgolt 
Started on the Finnish translation. Might be able to get it done this week.

boltgolt, 2023-02-28:
Thanks so much for the help @drcor @Edyburr @kuracode!
@BloodDrag0n tamil is currently at 0%, so i'd be very happy if you could help there

blackinitial, 2023-05-21:
I will continue the Indonesian translation

zhujinle, 2023-06-25:
Voted and polished for Simplified Chinese and Traditional Chinese translation.
Thanks for your nice tool and waiting for the release of new Howdy~
Is there any news about the release?

andywarh0l, 2024-02-25:
I'd like to translate it into Italian if someone hasn't already done so @boltgolt

joezhouchenye, 2024-03-12:
So, this is for the beta version with GUI support? For the master branch, I just need to modify the pam.py to translate into my language.

mvanzulli, 2024-03-16:
I can add Euskera if anyone already done it @boltgolt

Jhonata-Tirloni, 2024-10-04:
Hi @boltgolt , the link seens broken. I can help with brazilian portuguese and italian.

Bebegick, 2024-12-15:
Hello boltgolt,
I'm interested in contributing to this project, specifically by helping to fix/translate it into Indonesian. Could you please let me know if this issue is still open?

Thank you, and I look forward to your response!

IlNonoP, 2025-01-05:
I can't access to the link... can i contribute to italian?

TahierJerr, 2025-03-23:
I would like to help with Dutch!

binarydepth, 2025-07-14:
Do you have Spanish done?

[3 more replies]