boltgolt/howdy #487: Pubkey for Ubuntu GG not available.

paulwin1, 2020-12-31:
_Please describe the issue in as much detail as possible, including any errors and traces._
Adding the PPA repo on Ubuntu GG and Pop!_OS 20.10 results in a GPG error when updating.
```
paul@pop-os:~$ sudo apt update
[sudo] password for paul: 
Hit:1 http://apt.pop-os.org/proprietary groovy InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu groovy InRelease                     
Get:3 http://ppa.launchpad.net/boltgolt/howdy/ubuntu groovy InRelease [18.0 kB]
Hit:4 http://us.archive.ubuntu.com/ubuntu groovy-security InRelease
Hit:5 http://us.archive.ubuntu.com/ubuntu groovy-updates InRelease
Err:3 http://ppa.launchpad.net/boltgolt/howdy/ubuntu groovy InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8CEBC97F3E9E1D58
Hit:6 http://us.archive.ubuntu.com/ubuntu groovy-backports InRelease
Reading package lists... Done
W: GPG error: http://ppa.launchpad.net/boltgolt/howdy/ubuntu groovy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8CEBC97F3E9E1D58
E: The repository 'http://ppa.launchpad.net/boltgolt/howdy/ubuntu groovy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

----

I've searched for similar issues already, and my issue has not been reported yet.
Yes

Linux distribution (if applicable): Ubuntu 20.10 and Pop!_OS 20.10

Howdy version (`sudo howdy version`): cannot install Howdy.

boltgolt, 2020-12-31:
Seems like the key is not imported correctly, have your tried removing and adding the PPA?

paulwin1, 2020-12-31:
> Seems like the key is not imported correctly, have your tried removing and adding the PPA?

Yes I have. Multiple times.

Upon further inspection, there seems to be an issue with Pop!_OS and PPAs in general. I am unable to replicate this behavior in Ubuntu Budige, but can replicate similar issues with other PPAs on Pop!_OS. 

I will notify the Pop!_OS devs to verify if that is the issue. I will leave the issue open just in case. Thank you.

PolyPaul, 2021-01-23:
After speaking with the Pop devs it does seem like an internal issue with how pop handles PPAs in 20.10. There does not seem to be anything we can do to fix this error. Closing this until further notice from System76.

closed 2021-01-23 by PolyPaul