boltgolt/howdy #488: Howdy does not function, cannot find module dlib despite dlib being installed.

ccwky11, 2021-01-02:
As noted above, I cannot use howdy as when attempting to run howdy it fails as seen in the screenshot below
[image]
Any help or advice would be much appreciated, and thank you for this program, I have been using it since I began using Linux about 2 years ago and it is really one of the most impactful programs I have yet used.

ilkhem, 2021-01-02:
I also have this problem since the latest Manjaro update (it was working fine with the experimental 5.10 kernel, but broke once I upgraded to 5.10 LTS)

ccwky11, 2021-01-06:
Having attempted ilkhem's suggestion, I am still unable to get howdy working, I get the same error that no module is named dlib.

kxxt, 2021-01-16:
I also get this error after manjaro system upgrade.
I installed dlib, but it didn't solve the issue.
Now I'm trying to reinstall howdy.

catbabylon, 2021-04-06:
same issue on fedora 34 beta. I installed Howdy, then saw the dlib error as above, I installed dlib, same error message, I reinstalled howdy, same error message.

gabrieleaquaro, 2021-05-12:
In Fedora i  solved with :
`sudo dnf install python3-dlib`

closed 2026-11-16 by boltgolt