boltgolt/howdy #493: Add option to display Full Name in the "Identified face as ..." message.

PJSingh5000, 2021-01-10:
Currently, Howdy displays a message with the user's id:

    "Identified face as user_id"

I wold like to request the ability to make Howdy look up the user's name, and display that instead. For example:

    "Identified face as Mickey Mouse"

I understand that multiple user id's on a system could have the same full name. Ideally, this feature should be configurable system-wide for Howdy, for those that want to use it on their personal machines.

boltgolt, 2021-01-10:
I fully understand what you want, but I don't really get why this would be an improvement? I guess it would look slightly nicer in the console, but it would be a lot of work to setup an easily configurable system for this. If anybody is also interested in this leave a comment explaining why

PJSingh5000, 2021-01-10:
My motivation for this request was the login screen where the user's full name is displayed.

During authentication, the Howdy status message displaying the user's id seemed a little inconsistent when the user's full name is displayed in the login field.
(On some operating systems, the user doesn't ever get to see their user id).

When I suggested to make this "configurable" I was thinking of a simple key in `config.ini` (like `display_full_name`) with allowed values of `true` or `false`. Howdy would read this once during initialization. If true, Howdy would always look up the user's full name in order to format the "Identified face as ..." message.

There are a few options to get the user name.

One is to read and parse `/etc/passwd`...

    m_mouse:x:1000:1000:Mickey Mouse,,,:/home/m_mouse:/bin/bash

Another option is to use the `pwd` module in Python...

```
import pwd

pwd.getpwnam('m_mouse')
pwd.struct_passwd(pw_name='m_mouse', pw_passwd='x', pw_uid=1000, pw_gid=1000, pw_gecos='Mickey Mouse,,,', pw_dir='/home/m_mouse', pw_shell='/bin/bash')

pwd.getpwnam('m_mouse').pw_gecos.split(',')[0]
'Mickey Mouse'
```

boltgolt, 2021-01-11:
Ah sorry, I misunderstood you in your first post. I thought you meant custom defined names for each user in the models folder. Fetching the full name from the system makes much more sense and really should not be too hard. Probably enabling the setting by default