boltgolt/howdy #496: Whenever I boot up into 21.04, I get this error message: Sorry, the application has cli.py has stopped unexpectely.

goelvatsal, 2021-01-14:
I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Ubuntu 21.04 Beta

Howdy version (`sudo howdy version`):
Howdy 2.6.1

Whenever I boot up into 21.04, I get this error message: Sorry, the application has cli.py has stopped unexpectely.

[image]

boltgolt, 2021-01-19:
Weird, seems like `howdy test` is executed on bootup, have your made that a startup application?

goelvatsal, 2021-01-20:
No, I don't know how to set application to startup

dswij, 2021-03-04:
Can you confirm if `python3 /usr/local/bin/howdy test` command is in the startup applications list? (in Ubuntu, i believe it's https://help.ubuntu.com/stable/ubuntu-help/startup-applications.html.en)

goelvatsal, 2021-03-05:
Hi dswij, I reinstalled Ubuntu a while back and it appears to be working now. Thanks for your help! I'm going to close this thread.

closed 2021-03-05 by goelvatsal