boltgolt/howdy #497: Missing howdy

leoheck, 2021-01-16:
Howdy is missing from .deb and ubuntu ppa
Am I doing something wrong?

leoheck, 2021-01-16:
```
~ took 2s 
➜ sudo apt search howdy
Sorting... Done
Full Text Search... Done
howdy/groovy,groovy,now 2.6.1 all [installed]
  Howdy: Windows Hello style authentication for Linux.

~ 
➜ sudo apt install --reinstall howdy
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 38,9 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://ppa.launchpad.net/boltgolt/howdy/ubuntu groovy/main amd64 howdy all 2.6.1 [38,9 kB]
Fetched 38,9 kB in 1s (40,6 kB/s)
(Reading database ... 176068 files and directories currently installed.)
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Backup of Howdy config file created in /tmp/howdy_config_backup_v2.6.1.ini
Unpacking howdy (2.6.1) over (2.6.1) ...
Setting up howdy (2.6.1) ...
Processing triggers for man-db (2.9.3-2) ...

~ took 7s 
➜ sudo howdy                        
sudo: howdy: command not found
```

leoheck, 2021-01-16:
I see that the manual was installed.

boltgolt, 2021-01-17:
Whats your $PATH? If I remember correctly Howdy should be at `/usr/bin/howdy`, can you call that directly?

leoheck, 2021-01-17:
I am using the standard Ubuntu 20.04, I think my PATH is pretty standard.

[image]

The issue is more like the binary was not installed on the computer. Maybe it is not inside the package. Or maybe it has a different name before the installation.

See, there is nothing here called howdy.
[image]

boltgolt, 2021-01-17:
That's very weird, I've never seen this happen before. Have you tried reinstalling?

leoheck, 2021-01-17:
4 times, with .deb package and also with the ppa
Actually, I am running this on `20.10`

leoheck, 2021-01-17:
I will try once more later, do you have instructions for using it directly from this repo?

leoheck, 2021-01-19:
Wierd, I managed to make this work today using the PPA. Sorry for this report.

closed 2021-01-19 by leoheck

mksinicus, 2022-08-05:
Same issue here. I had the installation aborted and re-ran, got the same output and also that `sudo: howdy: command not found`. 

FYI I solved by removing Howdy and its dependencies and then installing it again under a better network condition.

xiaohuihuigh, 2023-02-15:
So what should I do when I got this issue?

mksinicus, 2023-02-15:
> So what should I do when I got this issue?

@xiaohuihuigh Try `sudo apt remove --purge howdy`, then reinstall it when you have stable access to `raw.githubusercontent.com`, where howdy tries to download its model files from (as far as I remember) (== 先卸载，挂个命令行里有效的梯子再安装)

xiaohuihuigh, 2023-02-15:
主要是那三个.bz2文件无法下载，导致没有安装成功，却没有报错。我从[其他地方](http://dlib.net/files/)下载好，放到新建的/usr/lib/security/howdy/dlib-data/文件夹下，再用apt安装，或者安装包安装就可以了。