boltgolt/howdy #498: AttributeError: 'VideoCapture' object has no attribute 'internal'

leoheck, 2021-01-19:
I am running this for the first time, not sure if I am doing this right.
I am seeing this issue

[image]

I am not sure which version I am using.  There is no way to get this info from the command line.
However, I am using ubuntu's PPA.
[image]

leoheck, 2021-01-19:
It looks like the config file was not filled with the default values.
The device_path was not right. 
What if you use the /dev/video0 as default instead of something that does not mean anything by default?

leoheck, 2021-01-19:
Found a way to get the version, but please, consider using the standard way for these things with (-v --version)
[image]

Also, why it has to open the camera to tell the version and then exit?

boltgolt, 2021-01-19:
The standard template for making an issue on GitHub literally tells you to run `sudo howdy version`, but you skipped using the bug template. It's also in the documentation and help text. 

If you actually read the error it says that the camera is configured incorrectly and to edit the device_path config option. 

`/dev/video0` is not a specific camera, it can change camera device on reboot because it's assigned by V4L. This would lead to a lot of big reports about randomly switching cameras. Use a by-path camera to keep using the same device 

It's opening the camera because you run it with sudo and howdy is enabled on your system. 

Please think before you post

closed 2021-01-19 by boltgolt

barbalex, 2023-11-13:
I am having the exact same problem.

Unfortunately the solution does not seem very clear to me.

How exactly can this problem be solved?

leoheck, 2023-11-13:
Oh, couldn't quite grasp it, so naturally, I just hit the delete button on my computer. Because, you know, typing a password like a regular human is just a breezeway easier and totally not a hassle at all.

barbalex, 2023-11-13:
I now found the solution here: https://github.com/boltgolt/howdy/issues/460#issuecomment-719905912
It seems that the device path is not set in Pop OS and has to be set via `sudo howdy config`