boltgolt/howdy #499: Test mode need some improvements.

leoheck, 2021-01-19:
Using version 2.6.1

The test version is not really good, I cannot increase the window. It does not show the text in the bottom of the window.
[image]

boltgolt, 2021-01-19:
Text at the bottom is just OpenCV debug text, doesn't really matter at all. The window will be as big as your webcam resolution so resizing is unnecessary. 

This is a lot of issues for just one person to open

closed 2021-01-19 by boltgolt

leoheck, 2021-01-19:
> This is a lot of issues for just one person to open
Yeap, it is necessary when the software is not working as described.

boltgolt, 2021-01-19:
Completely the wrong mindset when using a free, open source project.

leoheck, 2021-01-19:
> Completely the wrong mindset when using a free, open source project.

I can say the same to you when you said something like this `This is a lot of issues for just one person to open`

boltgolt, 2021-01-19:
If only you posted things that weren't posted in other issues already that would be fine. Fortunately I don't _have_ to deal with entitled people here