boltgolt/howdy #501: Archlinux package for pam-python still depends on python2

Bubbelb, 2021-01-20:
As I installed howdy on my Archlinux machine through AUR, I noticed a python2 dependency. Since python2 is deprecated, I just tried to change he dependency in the PKGBUILD to python. All went ok upon installation and in running. No problems whatsoever.

So I'd suggest to change the depencency from python2 to python.

boltgolt, 2021-01-20:
Probably a leftover from when pam-python only ran python2, thanks for the report

Bubbelb, 2021-01-21:
Oops, was too quick with my conclusion. I had python2 still installed when installing pam-python. So of course installation went ok.
Installation falis as python2 is not installed. At first I changed test.py and setup.py, so that they are running under python(3), not under python2. That doesn't help. The error now is::
````
==> Making package: pam-python 1.0.8-2 (do 21 jan 2021 17:13:01 CET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Removing existing $pkgdir/ directory...
==> Starting build()...
make --directory doc
make[1]: Entering directory '/home/bbl/Downloads/AUR/src/pam-python-1.0.8/doc'
sphinx-build -b html -E . html
Running Sphinx v3.4.3
loading intersphinx inventory from http://docs.python.org/2.7/objects.inv...
intersphinx inventory has moved: http://docs.python.org/2.7/objects.inv -> https://docs.python.org/2.7/objects.inv
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 1 source files that are out of date
updating environment: [new config] 1 added, 0 changed, 0 removed
reading sources... [100%] pam_python                                                                                                                                                           
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] pam_python                                                                                                                                                            
generating indices... genindex done
writing additional pages... search done
copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded.

The HTML pages are in html.
rm -f html/index.html && ln -s pam_python.html html/index.html
make[1]: Leaving directory '/home/bbl/Downloads/AUR/src/pam-python-1.0.8/doc'
make --directory src
make[1]: Entering directory '/home/bbl/Downloads/AUR/src/pam-python-1.0.8/src'
gcc -O0 -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -g -o ctest ctest.c -lpam
CFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -I/usr/local/lib/ " ./setup.py build --debug
/usr/lib/python3.9/site-packages/packaging/version.py:127: DeprecationWarning: Creating a LegacyVersion has been deprecated and will be removed in the next major release
  warnings.warn(
/usr/lib/python3.9/site-packages/packaging/version.py:127: DeprecationWarning: Creating a LegacyVersion has been deprecated and will be removed in the next major release
  warnings.warn(
running build
running build_ext
building 'pam_python' extension
creating build
creating build/temp.linux-x86_64-3.9
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fno-semantic-interposition -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -I/usr/local/lib/ -D_FORTIFY_SOURCE=2 -fPIC -g -DLIBPYTHON_SO="libpython3.9.so.1.0" -I/usr/include/python3.9 -c pam_python.c -o build/temp.linux-x86_64-3.9/pam_python.o
pam_python.c: In function ‘syslog_path_exception’:
pam_python.c:437:7: error: implicit declaration of function ‘PyClass_Check’; did you mean ‘PyFloat_Check’? [-Werror=implicit-function-declaration]
  437 |   if (PyClass_Check(ptype))
      |       ^~~~~~~~~~~~~
      |       PyFloat_Check
pam_python.c: In function ‘syslog_path_traceback’:
pam_python.c:581:5: error: ‘PyEval_CallObjectWithKeywords’ is deprecated [-Werror=deprecated-declarations]
  581 |     py_resultobj = PyEval_CallObject(pamHandle->print_exception, args);
      |     ^~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from pam_python.c:43:
/usr/include/python3.9/ceval.h:17:43: note: declared here
   17 | Py_DEPRECATED(3.9) PyAPI_FUNC(PyObject *) PyEval_CallObjectWithKeywords(
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
pam_python.c: In function ‘PamHandle_set_item’:
pam_python.c:879:11: error: assignment discards ‘const’ qualifier from pointer target type [-Werror=discarded-qualifiers]
  879 |     value = Py23_String_AsString(pyValue);
      |           ^
pam_python.c: In function ‘PamEnv_mp_assign’:
pam_python.c:107:26: error: implicit declaration of function ‘PyUnicode_Size’; did you mean ‘PyUnicode_Fill’? [-Werror=implicit-function-declaration]
  107 | #define Py23_String_Size PyUnicode_Size
[... 53 lines cut ...]
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:97,
                 from pam_python.c:43:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
   17 | Py_DEPRECATED(3.9) PyAPI_FUNC(PyObject *) PyEval_CallObjectWithKeywords(
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
error: command '/usr/bin/gcc' failed with exit code 1
make[1]: *** [Makefile:11: pam_python.so] Error 1
make[1]: Leaving directory '/home/bbl/Downloads/AUR/src/pam-python-1.0.8/src'
make: *** [Makefile:6: lib] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
````

I have no experience with Python, so I don't know what I can do about it...

BackMountainDevil, 2021-05-28:
wait for some cool guy to do this. I know some about python. Is there anything I can do?

iyanmv, 2022-09-24:
This is now more relevant because of [this](https://archlinux.org/news/removing-python2-from-the-repositories/).

BackMountainDevil, 2022-09-24:
That shoule be pam-python' Team to update its depends? Last time it's updated is 2020!

Pistrie, 2023-02-15:
You install install [this repository](https://github.com/FirefighterBlu3/python-pam) for `python-pam`. It is available in the AUR under `pam-python-git`

LudvigHz, 2024-01-06:
Is it a possibility to use [python-pam](https://github.com/FirefighterBlu3/python-pam) instead of pam-python? The former supports py3 and is packaged for arch linux which would simplify the installation of howdy greatly. At the moment, arch users need to either use the pam-python fork or build python2 from the AUR, which is a PITA.

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against Howdy v2. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt

Bubbelb, 2026-11-16:
Agreed. V3 is the way to go. Thank you.