boltgolt/howdy #503: Option to not allow facial recognition on first login

gidoBOSSftw5731, 2021-01-25:
I like the idea of this, but am concerned of the implications if it could be used for first login. I'd be more comfortable if it could be turned off on the login (not lockscreen) page, and/or if it could be disabled with a macro until next successful login.

Cheers, and nice job figuring this out decently!

boltgolt, 2021-01-25:
Unfortunately Howdy can't know WHY it's being called, so it's hard to skip first login. However with [this wikipage](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types), you could enable it only for sudo, for example.

boltgolt, 2021-01-25:
Additionally you could use `sudo howdy disable` to write a macro for it if you wanr to go that route

closed 2021-01-25 by boltgolt