boltgolt/howdy #507: Does not detect face when inverted

thiagoricciardi, 2021-02-11:
Dear,

First let me thank you all for the wonderful work and excellent job creating this piece of gold :smiley: 

I daily use my laptop (a convertible 2-in-1) almost all the time in an upside-down (inverted) position, so it frustrates me that I cannot get howdy to detect my face when it's inverted.
I cannot even create a model when the camera (or in this case, the laptop) is upside-down, howdy does not even detect a face when in this position.
When the laptop (and consequently the camera) is resumed to it's "normal" orientation, howdy works fine.

Is this intended behavior? Shouldn't the AI be able to identify objects (or in this case, my face :stuck_out_tongue_closed_eyes:) regardless of their orientation?

Thanks a lot!

Best regards!

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04

Howdy version (`sudo howdy version`): Howdy 2.6.1

boltgolt, 2021-02-11:
To optimize recognition time the face location algorithm has only been trained on faces that are the right side up. This makes sense as in almost all cases the face will be in that orientation and looking for an inverted face is unneeded. Unfortunately retraining this model is not an option.

I could add an configuration option that makes howdy scan both a normal frame and an 180 degree rotated frame. If enough people need this i will consider adding it.

thiagoricciardi, 2021-02-11:
Hi!
Thanks a lot for the explanation and quick response!

In my opinion, the most elegant solution is for the system to correct the orientation of the camera (you see, when I'm in conference calls I cannot use the laptop upside-down because of this), there are sensors in the laptop for it to work, but I'm giving a break on this path for now, I could not find anything online that could help me change the camera orientation based on internal sensors easily.
So your solution looks very promising for me, and it's very nice too! I do have some development skills and, with the correct guidance, I could try to make a pull request with the modifications you suggested if somebody are willing to point me in the right direction.

Thanks again!
Best regards!

boltgolt, 2021-02-15:
I think that's a much more elegant solution, rotating all cameras based on system orientation sensors. Did you check if [iio-sensor-proxy](https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/) can detect your gyroscope?

Kira-PH, 2026-11-16:
My ask (https://github.com/boltgolt/howdy/issues/415) was for it to be configurable. *But* that was six years ago, and I'm not using Howdy at present.