boltgolt/howdy #510: Install Howdy on Arch linux

kushmittal2009, 2021-02-23:
@boltgolt for you said that a maintainer is requred fro arch linux howdy, it is very easy to install howdy on linux as well. First download the dep file and in arch linux terminal you could cd into the directory you downloaded it.

Download the .deb file from the Releases page of howdy

Next cd in to downloads

cd ~/Downloads

Next install the deb with dpkg -i

dpkg -i

sudo dpkg -i howdy_2.6.1.deb 

//Replace howdy_2.6.1.deb with package name of downloaded howdy

boltgolt, 2021-02-23:
Hey @kushmittal2009, thanks for the quick guide! Are you sure you are using [Arch Linux](https://en.m.wikipedia.org/wiki/Arch_Linux)? It looks like you are using a `dpkg` based distribution such as Debian or Ubuntu.

kushmittal2009, 2021-02-23:
@boltgolt I could try using a live usb to make sure I have not installed any other package and try it

Jellyfrog, 2021-03-12:
Howdy is already available on Arch, https://aur.archlinux.org/packages/howdy/

boltgolt, 2021-03-12:
It is, I think this is a good attempt at helping the project though

closed 2021-03-12 by boltgolt

Jellyfrog, 2021-03-12:
@boltgolt speaking off, I could step in as maintainer. Would be my first package but how-hard-can-it-be

SnakeXrnt, 2025-10-07:
basically, howdy guide is having a bit of a problem for me, so i did my own guide and here how i do it myself

install howdy from yay. or other things, just make sure you can do sudo howdy add 

sometime your emmiter wont work, at that point just try to use linux-ir-emmiter-enable or something like that in howdy guide

and the wrong point is the path for howdy pam. try to search where is your pam by doing this
```sudo find / -path '*/security/*howdy*' 2>/dev/null```

after you find the correct path for howdy pam, you are going to put it in your pam.d 

go to /etc/pam.d/ , and just `ls -la` there, you can see what kinda thing you can put face recog on. maybe you want to edit sudo, use it like this 

```
#%PAM-1.0 (this one is already there, put this 2 lines below this pam-1.0)
auth      optional      pam_exec.so /usr/bin/linux-enable-ir-emitter run
auth      sufficient    (your howdy pam path) timeout=4
```

then just setup and configure howdy. and you will be done

# Linux-enable-ir-emmiter, (problematic AUR)

this one is a bit tedious if you use hyprland, you dont have qt software in hyprland. so basically you have to configure it like this

`sudo linux-enable-ir-emitter configure --no-gui`

after that, the emmiter should work fine even after you reboot.

SnakeXrnt, 2025-10-07:
basically, howdy guide is having a bit of a problem for me, so i did my own guide and here how i do it myself

install howdy from yay. or other things, just make sure you can do sudo howdy add 

sometime your emmiter wont work, at that point just try to use linux-ir-emmiter-enable or something like that in howdy guide

and the wrong point is the path for howdy pam. try to search where is your pam by doing this
```sudo find / -path '*/security/*howdy*' 2>/dev/null```

after you find the correct path for howdy pam, you are going to put it in your pam.d 

go to /etc/pam.d/ , and just `ls -la` there, you can see what kinda thing you can put face recog on. maybe you want to edit sudo, use it like this 

```
#%PAM-1.0 (this one is already there, put this 2 lines below this pam-1.0)
auth      optional      pam_exec.so /usr/bin/linux-enable-ir-emitter run
auth      sufficient    (your howdy pam path) timeout=4
```

then just setup and configure howdy. and you will be done

# Linux-enable-ir-emmiter, ( AUR)

this one is a bit tedious if you use hyprland, you dont have qt software in hyprland to give the gui. so basically you have to configure it like this

`sudo linux-enable-ir-emitter configure --no-gui`

after that, the emmiter should work fine even after you reboot.